lancaster-university/codal-microbit-v2 #113: Should uBit.logo work when switched to TouchMode::Resistive?

martinwork, 2021-05-29:
Does uBit.logo work only in capacitative mode or does the code below use the wrong method to switch modes?

After pressing button A to select TouchMode::Resistive, the logo is always pressed.

This example is intended to match the MakeCode code in https://github.com/microsoft/pxt-microbit/issues/3724.

[test-pintouch-v2.zip](https://github.com/lancaster-university/codal-microbit-v2/files/6564661/test-pintouch-v2.zip)

```
#include "MicroBit.h"

MicroBit uBit;

void onButtonA(MicroBitEvent e)
{
    uBit.io.logo.isTouched( codal::TouchMode::Resistive);
}

void onButtonB(MicroBitEvent e)
{
    uBit.io.logo.isTouched( codal::TouchMode::Capacitative);
}

void display()
{
   while (true)
   {
     uBit.display.printChar( uBit.logo.isPressed() ? '1' : '0');
     uBit.sleep(100);
   }
}

int  main() {
    uBit.init();

    uBit.messageBus.listen( MICROBIT_ID_BUTTON_A,  MICROBIT_BUTTON_EVT_CLICK, onButtonA);
    uBit.messageBus.listen( MICROBIT_ID_BUTTON_B,  MICROBIT_BUTTON_EVT_CLICK, onButtonB);

    create_fiber( display);

    release_fiber();
}
```

microbit-carlos, 2022-05-05:
I can confirm that In this example `uBit.logo.isPressed()` does not work as expected.

`uBit.io.logo.isTouched()` does work, even when called without setting a TouchMode argument (as it is already changed when button A or B are pressed). 

The same can be replicated in MakeCode, as it uses `uBit.logo.isPressed()`: [microbit-logo-touch-change.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/8633142/microbit-logo-touch-change.hex.zip)

[image]

```typescript
input.onButtonPressed(Button.A, function () {
    basic.showString("r")
    pins.touchSetMode(TouchTarget.LOGO, TouchTargetMode.Resistive)
})
input.onButtonPressed(Button.B, function () {
    basic.showString("c")
    pins.touchSetMode(TouchTarget.LOGO, TouchTargetMode.Capacitive)
})
basic.forever(function () {
    if (input.logoIsPressed()) {
        basic.showIcon(IconNames.Heart)
    } else {
        basic.showIcon(IconNames.Sad)
    }
})

```

martinwork, 2022-12-31:
MakeCode block "set logo to touch mode %mode" uses uBit.io.logo.isTouched(mode), but I don’t think uBit.logo supports changing the uBit.io.logo touch mode.

uBit.logo is a TouchButton, whose NRF52Pin (uBit.io.logo) is initially set as output and monitored via GPIOTE events in NRF52TouchSensor (uBit.touchSensor). uBit.logo is not aware of changes to uBit.io.logo via NRF52Pin.

Changing mode with uBit.io.logo.isTouched(Resistive) creates a Button, sets the pin to input, and sets its pull mode, which presumably clashes with the ongoing uBit.logo TouchButton and NRF52TouchSensor GPIOTE code.

Changing mode with uBit.io.logo.isTouched(Capacitive) adds a second TouchButton to uBit.touchSensor, and sets the pin as output again. After that, uBit.logo.isPressed() seems to start working again, though I think it no longer has the “empty timeslot for that channel to drain all its charge before sampling it again”, and MakeCode’s “on logo pressed” block sees double events.

Do we need resistive touch for the logo?

martinwork, 2022-12-31:
The Python program below also doesn't work after changing to resistive mode, and the logo is_touched() state always seems less stable than MakeCode, I guess because microbit_hal_pin_is_touched uses uBit.logo.buttonActive(), which misses out on the isPressed() debounced state.

```
from microbit import *

while True:
    if pin_logo.is_touched():
        display.show(Image.HAPPY)
    else:
        display.show(Image.SAD)
    if button_a.was_pressed():
        pin_logo.set_touch_mode(pin_logo.RESISTIVE)
    if button_b.was_pressed():
        pin_logo.set_touch_mode(pin_logo.CAPACITIVE)
```

microbit-carlos, 2023-01-06:
Thanks Martin, I can confirm that with the MicroPython example `pin_logo` when set to resistive is always triggered as "touched".

microbit-carlos, 2024-03-05:
Quick summary:
- uBit.logo default touch mode should be capacitive
- uBit.io.logo default is not clear, might be either, although ideally it should be capacitive as well
- Touch for pins default should be resistive (for V1 compatibility)
- Changing the logo to resistive touch should work (the main problem described in this issue)

microbit-carlos, 2024-03-26:
Okay, a bit more detail:

> * uBit.logo default touch mode should be capacitive

This is correct right now, no action required.

> * uBit.io.logo default is not clear, might be either, although ideally it should be capacitive as well

uBit.io.logo default is resistive, and it should be capacitive:
- https://github.com/lancaster-university/codal-microbit-v2/issues/411

> * Touch for pins default should be resistive (for V1 compatibility)

This is also correct right now and no action is required.

> * Changing the logo to resistive touch should work (the main problem described in this issue)

This is the topic for this issue.

Additionally
- https://github.com/lancaster-university/codal-microbit-v2/issues/416

microbit-carlos, 2024-03-26:
Conclusion, we'll go for simplest fix:
- MakeCode to move to uBit.io.logo
- uBit.init() to set uBit.io.logo to capacitive mode

microbit-carlos, 2024-03-27:
PR to set uBit.io.logo to capacitive by default:
- https://github.com/lancaster-university/codal-microbit-v2/pull/418

Once that it's merged we can submit this to MakeCode, I've tested this with pxt locally and works in resistive mode:
- https://github.com/microsoft/pxt-microbit/compare/master...microbit-carlos:pxt-microbit:logo-pin?expand=1

microbit-carlos, 2024-07-26:
Raised PR:
- https://github.com/microsoft/pxt-microbit/pull/5749

We can close this ticket now.

closed 2024-07-26 by microbit-carlos