lancaster-university/codal-microbit-v2 #115: P0 turns into P1 after using uBit.audio.setPin(uBit.io.P1)

dpgeorge, 2021-06-07:
See https://github.com/microbit-foundation/micropython-microbit-v2/issues/69#issuecomment-855948709

Reproduction in C++ using CODAL v0.2.28:
```c
#include "MicroBit.h"

MicroBit uBit;

int main() {
    uBit.init();

    uBit.serial.printf("== start ==\r\n");
    uBit.serial.printf("P0.name=%d\r\n", uBit.io.P0.name);
    uBit.serial.printf("P1.name=%d\r\n", uBit.io.P1.name);
    uBit.serial.printf("P2.name=%d\r\n", uBit.io.P2.name);

    uBit.audio.setPin(uBit.io.P1);

    uBit.serial.printf("== end ==\r\n");
    uBit.serial.printf("P0.name=%d\r\n", uBit.io.P0.name);
    uBit.serial.printf("P1.name=%d\r\n", uBit.io.P1.name);
    uBit.serial.printf("P2.name=%d\r\n", uBit.io.P2.name);

    for(;;);
}
```
Output:
```
== start ==
P0.name=2
P1.name=3
P2.name=4
== end ==
P0.name=3
P1.name=3
P2.name=4
```
From then on, P0 behaves like P1 because its `.name` field has been overwritten with "3".

martinwork, 2021-06-07:
Thanks @dpgeorge . I can see why: MicroBitAudio::pin is a NRF52Pin & reference.

@finneyj Should I simply change the internal MicroBitAudio::pin to a pointer?

finneyj, 2021-06-07:
Yes please @martinwork - absolutely the thing to do.

Apologies @dpgeorge.

martinwork, 2021-06-07:
Created PR https://github.com/lancaster-university/codal-microbit-v2/pull/116.

closed 2021-06-07 by martinwork

dpgeorge, 2021-06-08:
Thanks for the quick turn around!  Can confirm the bug is fixed.

martinwork, 2021-06-08:
Thanks for confirming @dpgeorge