lancaster-university/codal-microbit-v2 #120: datalog: Add "data preview" button with graph

jaustin, 2021-06-14:
We would like to add a 'data preview' button to the (online) HTML page that shows the user a graph of their data.

This is a limited scope feature (we don't want to encourage the idea that you can use this instead of Excel/Sheets etc). It should signal that if you want more detailed graphing you should use another tool. For example.

* New button "Data preview" ("Preview graph"?)
* When you click this button, it:
  * adds a message to the page that says "This is a visual preview of your data. For more advanced graphs or to analyse the data before graphing, please download the CSV and open it in a spreadsheet and graphing tool" (@microbit-giles @microbit-mark are you happy with this sentence?)
  * Adds a graph to the page that fills the width of the page and is above the table.

The graph
* Should have labelled axes (Time (use label from timestamp column) / Value)
* Should assume timestamps, if these aren't present, we can fail with a message "please ensure timestamps are enabled to enable the data preview"
* Should use microbit brand colours for each data series, with a legend (@microbit-matt-hillsdon to come back with whether data markers can mitigate any accesibilty concerns with the colours)
* Each data series should have a label that equates to the column header in the legend
* The graph should, by default, cover the whole series of data but be zoomable
* We should skip additional 'header' rows. It would be acceptable NOT to introduce a new data series that is added part way through a log file, (IE only graph data series that are present at the top of the log)
* The user does not need to be able to download the graph, just the data (but right click and export image if it comes for free with the library is fine)

Questions
* Do we need to mitigate CPU usage/loading times for long log files, or provide a 'loading' screen when the file is full?
* Does loading the graph stop the autorelaod of the page when there is new data, or do we reload with a parameter to say we want the graph to be visible?
 * (If not, we probably do need to signal that the data was reloaded so the user knows why their graph disappeared)

jaustin, 2021-06-14:
@microbit-carlos FYI, thanks for the PoC :D

microbit-carlos, 2021-06-14:
My assumption would be that the "Data preview" graph feature would be only available if the file is viewed online, and that the offline experience would should an explanation. Would that be correct?

If this is the case @microbit-giles made a good point in a call last week about being able to clarify that the data is in the micro:bit, not in a micro:bit website or server. This would be a very understandable misconception if we use an HTML file and some of the features are only available if the user is online.

microbit-giles, 2021-06-14:
> adds a message to the page that says "This is a visual preview of your data. For more advanced graphs or to analyse the data before graphing, please download the CSV and open it in a spreadsheet and graphing tool" 

Some wording like this is an excellent idea. I think we should bear in mind the copy and paste route as well as CSV download, as well as clarifying where the data is.

How about something like this? "This is a visual preview of the data on your micro:bit. To analyse it in more detail or create your own graphs, transfer it to your computer. You can copy and paste your data, or download it as a CSV file which you can import into a spreadsheet or graphing tool."

jaustin, 2021-06-14:
>"This is a visual preview of the data on your micro:bit. To analyse it in more detail or create your own graphs, transfer it to your computer. You can copy and paste your data, or download it as a CSV file which you can import into a spreadsheet or graphing tool."

That works for me! Nicely put.

jaustin, 2021-06-14:
> My assumption would be that the "Data preview" graph feature would be only available if the file is viewed online, and that the offline experience would should an explanation. Would that be correct?
> 

I think for compactness the offline experience would just omit the button.

> If this is the case @microbit-giles made a good point in a call last week about being able to clarify that the data is in the micro:bit, not in a micro:bit website or server. This would be a very [...]

Right... but do we need to say that in the file, or just in the docs.

A simple way to clarify would be to have the data table say above it this slightly modified version of Giles' suggestion above, then switch the text out for the visual preview (in online mode).

"This is the data on your micro:bit. To analyse it and create your own graphs, transfer it to your computer. You can copy and paste your data, or download it as a CSV file which you can import into a spreadsheet or graphing tool.

microbit-carlos, 2021-06-14:
> I think for compactness the offline experience would just omit the button.

This might be confusing for resources that mention the button and then an offline users doesn't find it in the HTML.
I would assume most people would never encounter the offline version, so this might catch by surprise many people writing resources.

> Right... but do we need to say that in the file, or just in the docs.

That's a good question. The docs might the easiest place to explain this, but I would bet most misconceptions come around precisely for not reading the docs.

> A simple way to clarify would be to have the data table say above it this slightly modified version of Giles' suggestion above, then switch the text out for the visual preview (in online mode). [...]

Right, so this text would replace the current `This table shows data logged from your micro:bit. <a ...>Learn about the data logging feature</a>.`?

I think the main point of confusion would be if we keep the "Data preview" button, because at that point in offline mode it needs to clarify that this is an "online only" feature, and that's when people might wonder if the data is being sent somewhere.

jaustin, 2021-06-15:
In our discussion this morning:
* We think it's okay to not show the button in offline mode (because showing it requires a bunch of 'large' messages about why the feature isn't available offline when the data is not sent to any servers, etc), so we should omit the button
* If we have the bytes, it might be good to warn the user they're in a limited mode, something like  
 * "Offline: limited features" 
 * "Offline: no graphing"
 * "Offline: basic mode"
 * "Could not load features from web"
 * "No internet connection"
 * "Offline"

@microbit-giles could you propose some better wording or pick one of those?
@microbit-matt-hillsdon do we have the bytes for any of these? (if we're not doing the data preview button)

microbit-giles, 2021-06-15:
If the main difference in functionality is lack of graphing, I think 'Offline: no graphing' is good, but words should probably match the (missing) button in case they are following a screengrab / guide. eg if button is 'Data preview' it should day 'Offline: no data preview' or if there's room, 'You're offline, no data preview available'

closed 2021-06-22 by finneyj

microbit-matt-hillsdon, 2021-06-22:
I'll update the data logging support files on microbit.org later this evening.

finneyj, 2021-06-22:
cool. :)