lancaster-university/codal-microbit-v2 #123: datalog: hide/remove MY_DATA.HTM when data logging activity is not taking place

microbit-mark, 2021-06-15:
Once MY_DATA.HTM exists on the MICROBIT drive it does not seem that it can be removed. You still have a MICROBIT drive that shows a MY_DATA file with nothing in it even if you flash a new program  that does not contain data logging APIs.

In a classroom environment with shared micro:bits, this might be confusing/worrying to people who are handed a micro:bit with an empty file on it called MY_DATA, that when opened exposes some buttons and a link to microbit.org.

is it possible to hide/remove the file if a program is flashed that does not use the data logging APIs.
a
The MakeCode editor has an option to 'full' erase the data and I would expect this to have the same effect of removing the file.

Another option would be to add some indicator that the file is only useable/populated when using the data logging feature or if no data is found, link people directly to microbit.org or to this article on the MY_DATA file https://support.microbit.org/en/support/solutions/articles/19000127398-my-data-htm

microbit-mark, 2021-07-01:
@finneyj I think the outcome of our discussion about this that we would hide the file on 'full' erase? Is that your understanding too?

finneyj, 2021-07-01:
Yes - I think we've already applied that here: 
https://github.com/lancaster-university/codal-microbit-v2/commit/fb96113f8549f0ea0627e0e9ad14fa58b370402f

microbit-carlos, 2021-07-02:
Tested and can confirm it works 👍 
Thanks Joe!

closed 2021-07-02 by microbit-carlos