lancaster-university/codal-microbit-v2 #124: daalog: odd characters in data log mean page does not update when micro:bit is dis/reconnected

microbit-mark, 2021-06-17:
Using a program in MakeCode
[image]

1. Open MY_DATA page in browser and view logs. Initial results are seen
2. Run program until log full
3. View log, which shows 'LOG FULL'
[image]
4. Disconnect and reconnect micro:bit. Page does not refresh. manual refresh does not update data
5. View source. '?' characters at point where the page no longer presents log '16.5' seconds
[image]
6.  Scroll to end of source. All results are visible in the source
[image]

Windows 10
Edge 91
micro:bit RC1, DAPLink 0255

microbit-mark, 2021-06-17:
Similar issue reported where dis/reconnect does not update data, but unpairing the micro:bit from the browser and then refreshing the page resolves the issue. @microbit-aimee can you add your reproducer here too and we can see if it's related.

microbit-aimee, 2021-06-17:
> Similar issue reported where dis/reconnect does not update data, but unpairing the micro:bit from the browser and then refreshing the page resolves the issue. @microbit-aimee can you add your reproducer here too and we can see if it's related.

This was the data logged yesterday - first entry was an isolated log, thereafter data did not update until micro:bit was unpaired. Then a refresh of page delivered the new data.
[microbit (5).csv](https://github.com/lancaster-university/codal-microbit-v2/files/6670857/microbit.5.csv)

finneyj, 2021-06-17:
Thanks @microbit-mark  @microbit-aimee  

I'm just trying to understand this a bit more. When you say "unpair"  do you mean unplugging the micro:bit from the USB port and then plugging it back in again?

Does closing the browser and reopening the MY_DATA.HTM always resolve the issue?

(p.s. Hi Aimee!)

finneyj, 2021-06-17:
Ah, right, gotcha.

There's nothing WebUSB related in CODAL (that's all MakeCode side), so yes, probably something to raise over there. Happy to host the issue here until resolved though of course!

microbit-mark, 2021-06-18:
@finneyj Raised in https://github.com/microsoft/pxt-microbit/issues/4203 too. I'm struggling to get a consistent reproducer, but there's defintely a bug around the data not updating on the page when the micro:bit is disconnected and reconnected. The second time I tried to reproduce, the buttons and the table did not load at all.

[image]

finneyj, 2021-06-18:
That data does look very weird @microbit-mark 

Can you send me a copy of that MY_DATA.HTM file?

microbit-mark, 2021-06-18:
[MY_DATA.HTM.zip](https://github.com/lancaster-university/codal-microbit-v2/files/6676346/MY_DATA.HTM.zip)

I just reloaded the page again and got this

[image]

finneyj, 2021-06-18:
Thanks @microbit-mark. 

Here's a view in a HEX editor:

[image]

You can see from this that there is a well defined pattern here. The areas containing "FF" are unwritten FLASH areas, right in the middle of the HTML header... so something definitely afoot here.

I have seen this once before. It was caused by the KL27 (USB interface chip where the data is stored) rejecting/failing instruction from the micro:bit to write data. I think this happens primarily when the KL27 is busy doing other things (e.g. managing the USB connection).

I put in some robust detection for this and retry operations when it failed, which solved the problem in all my tests... but I've never tested with WebUSB active... it could be that the WebUSB interface is putting a high demand on the KL27, and it doesn't have time to record the data...

finneyj, 2021-07-27:
I added some additional diagnostics I think I've tracked this back to I2C transaction requests (particularly FLASH write operations) being lost when WebUSB is in use with MakeCode. This appears to be caused by the KL27 interface chip issuing out-of-date responses to I2C requests (presumably when under higher CPU load due to the WebUSB activity).

There's a very clear trend that shows it is possible to read back the result from a previous transaction even though a new one has been issued and received (at some level) by the KL27. I suspect what is happening here is that:

The KL27 is receiving the request and leaving it in a buffer while it does other things (e.g. USB activity)
The IRQ line is shared, so false positive interrupts are common, so the NRF52 may solicit a response before the request is fully processed. When it does so, it receives an out of date response (whatever happens to be in the buffer from the last transaction)
This is particularly problematic if the NRF52 issues consecutive commands with the same command code, as it is then impossible to distinguish a successful completion from an "in progress" condition. Such consecutive transactions are very common in our key data logging use case with the FLASH_WRITE command for example, and similarly with a FLASH_ERASE command when erasing/formatting the file system.

I tested this theory by interspersing every transaction with another (pointless but different) transaction such as GET_FLASH_SECTOR_SIZE, in order to separate out the two conditions. This appeared to resolve the issue at the expense of performance. 

I've introduce that workaround into a patch included the v0.2.33 release. This seems to improve stability in my tests. 

I suspect we'll see reports of logging performance issues  when connected to MakeCode using WebUSB however, as the logging speed is visibly reduced when WebUSB is in use.

Work is in train to fix this in the next release of DAPLink.

microbit-mark, 2021-08-03:
Tested the [original reproducer](https://github.com/lancaster-university/codal-microbit-v2/issues/124#issue-923908381) in https://makecode.microbit.org/beta#editor which is tracking CODAL v0.2.33 and this no longer reproduces. 🎉 

Attaching MY_DATA file which shows no corrupt data
[MY_DATA.HTM.zip](https://github.com/lancaster-university/codal-microbit-v2/files/6922470/MY_DATA.HTM.zip)
[microbit.csv](https://github.com/lancaster-university/codal-microbit-v2/files/6922479/microbit.csv)

There appears to be a small glitch in the CSV at line 3491 where the micro:bit has been dis/reconnected and the time stamp is not logged and the accel. values remain on the same line
[image]

At this point, I wanted to leave it logging and look at another bug so I switched tabs in Chrome to another MakeCode instance, which had the effect of claiming the WebUSB interface and stopping the data logging. I don't expect many people will come across this, but I'll raise this for discussion with MakeCode.

microbit-mark, 2021-08-03:
Been playing a bit more with this and I think it's still buggy if you interact with the device console in MakeCode
[image]

https://makecode.microbit.org/_R4whXsaEFVqC

[MY_DATA.HTM.zip](https://github.com/lancaster-university/codal-microbit-v2/files/6923688/MY_DATA.HTM.zip)

It seems if I select the device console window and change between this and blocks it leaves erroneous data in the log.

finneyj, 2021-08-03:
Thanks @microbit-mark 

Erroneous how? The data log seems to look consistent apart from the last (incomplete) entry?

microbit-mark, 2021-08-03:
Perhaps it's fine. The html view shows data down to 18.12 seconds and does not update after that. At this point in the source view, we see `????` in the source
[image]
and then later on  
[image]

[image]

I think this may be where I am switching to device console view in MakeCode

finneyj, 2021-08-03:
Hmm, yeah that does look weird. The journal metadata also still looks a bit off. There should just be one number in here, with the rest zeroed out as it becomes obsolete...

[image]

microbit-mark, 2021-08-20:
I can't seem to reproduce the bug in https://github.com/lancaster-university/codal-microbit-v2/issues/124#issuecomment-891771128 now. Tested a few time today in makecode beta.

microbit-mark, 2021-09-08:
As discussed on MakeCode call, created program to log a value every 50ms [microbit-log-forever-50ms.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/7127224/microbit-log-forever-50ms.hex.zip) using makecode.microbit.org/beta

I've been running this for half an hour without issue

microbit-carlos, 2022-05-23:
This should as discussed in this thread should have been fixed with the busy flag implemented in V2.2 release 0257 and 0258 beta test builds of V2.00.

Any problems similar to this one that are still happening should be captured in a new issue.