lancaster-university/codal-microbit-v2 #134: BLE DFU Flash causes drop into bootloader after reset

martinwork, 2021-06-25:
When a new hex has been BLE DFU flashed, pressing reset enters the bootloader, and a plus sign is displayed. This seems to be due to changes since v0.2.25.

martinwork, 2021-06-25:
Adding a return at the top of eraseUserStorage removes the problem.
https://github.com/lancaster-university/codal-microbit-v2/blob/master/model/MicroBit.cpp#L365

martinwork, 2021-06-25:
It looks like a DFU is choosing a CRC check on the app, and this is persisting after the DFU.
https://github.com/microbit-foundation/v2-bootloader/blob/master/nRF5SDK_mods/components/libraries/bootloader/dfu/nrf_dfu_validation.c#L217

I guess I thought a reset wouldn't run the CRC because of this flag
https://github.com/microbit-foundation/v2-bootloader/blob/master/bootloader/microbit/config/sdk_config.h#L73

finneyj, 2021-06-25:
Good catch @martinwork!  Thanks for tracking this down.

Hmm.... this is a juicy one. I don't think we really want to bypass the formatting of the log when a new program is flashed, as that's the intentional behaviour we decided we wanted. If we don't clear this persistent flag, then it will reformat the KL27 flash every time the device resets - and that would be very bad for flash wear.

Do we know exactly when the bootloader checks the CRC?  Is it on every reset?

martinwork, 2021-06-25:
I didn't think the CRC would run for a reset because of the NRF_BL_APP_CRC_CHECK_SKIPPED_ON_SYSTEMOFF_RESET, but clearly it does. The SDK code is:
```
static bool crc_on_valid_app_required(void)
{
    bool ret = true;
    if (NRF_BL_APP_CRC_CHECK_SKIPPED_ON_SYSTEMOFF_RESET &&
        (nrf_power_resetreas_get() & NRF_POWER_RESETREAS_OFF_MASK))
    {
        nrf_power_resetreas_clear(NRF_POWER_RESETREAS_OFF_MASK);
        ret = false;
    }
```

I have created https://github.com/microbit-foundation/v2-bootloader/pull/17. This is one way to remove the problem for future hexes.

BLE flashing over an old hex will still have the problem. I think the apps can overcome that by using a full Nordic init packet instead of the simple one.

microbit-carlos, 2024-05-07:
PR https://github.com/microbit-foundation/v2-bootloader/pull/17 was merged into the bootloader and released in v1.1.0 https://github.com/microbit-foundation/v2-bootloader/releases/tag/v1.1.0, which was included in codal-microbit-v2 in https://github.com/lancaster-university/codal-microbit-v2/commit/30bdc29d75610988a50bb6de6514fbd1c2b5c5b1.

@martinwork can this issue be closed as completed? or does it include other things that still need to be addressed.

martinwork, 2024-05-07:
@microbit-carlos Yes, thanks. As far as I am aware, the bootloader update fixed it.

closed 2024-05-07 by martinwork