lancaster-university/codal-microbit-v2 #136: datalog: serial mirroring new line doesn't have a return carriage

microbit-carlos, 2021-07-06:
During the MicroPython implementation @dpgeorge noticed that the serial mirroring only does a `\n` instead of `\r\n`. Should we update this in CODAL?

finneyj, 2021-07-06:
Hi @microbit-carlos 

My, this is an age old fight to have. :)

I personally have no strong feelings either way. Happy to emit CRLF if there's a consensus, but would be good to get some input from the MakeCode team here too, as they're close to a release...

@abchatra @jwunderl - Any thoughts here?

abchatra, 2021-07-06:
@jwunderl what does serial.writeLine emit?

jwunderl, 2021-07-06:
we default to `\r\n` https://github.com/microsoft/pxt-microbit/blob/master/libs/core/serial.ts#L35

finneyj, 2021-07-06:
Thanks @jwunderl  @abchatra.

So it sounds like changing the serial mirroring to \r\n wouldn't keep you guys awake at night?

jwunderl, 2021-07-06:
Sounds fine to me~

finneyj, 2021-07-27:
Fixed in https://github.com/lancaster-university/codal-microbit-v2/commit/2f2cbfa4fb6b0f1d46b3849362e78ad366e18c40

I've left the data logged to FLASH unchanged though - that is still terminated with '\n' only, as introducing '\r\n' in there would reduce the amount of usable storage for kids an teachers quite noticeably if you think about it. A single column log recording a typical three digit number would see a 20% reduction in available storage. which doesn't seem like a good tradeoff since the JS on the webpage parses it out anyway...