lancaster-university/codal-microbit-v2 #138: How can a program get the CODAL version?

martinwork, 2021-07-29:
microbit_dal_version() is declared but not implemented.
MICROBIT_DAL_VERSION is given a default value in config.h but not set to a useful version in the build.

martinwork, 2021-07-29:
There's CodalDevice::getVersion() which returns DEVICE_DAL_VERSION, but that also seems to be "unknown".

microbit-carlos, 2023-03-02:
This is needed for the next firmata release, so we need to prioritise it.

microbit-carlos, 2023-03-06:
@JohnVidler once this is implemented we need to add it to test it in the firmata project:
- https://github.com/microbit-foundation/microbit-firmata/issues/16

microbit-carlos, 2024-03-11:
Should be fixed with these two PRs:
- https://github.com/lancaster-university/microbit-v2-samples/pull/87
- https://github.com/lancaster-university/codal-microbit-v2/pull/413

The only disadvantage is that this would not work when the codal.json file has `target.dev` set.

JohnVidler, 2024-03-26:
I've prototyped a better CODAL_VERSION_* definition set over in `feature/versionable` in `codal-core`: https://github.com/lancaster-university/codal-core/compare/master...feature/versionable

This should check the tag/branch/hash version of the _target_ rather than core itself, so we should get sensible values back.

If this is merged into master, we can then use the generated definitions in `microbit_dal_version()` and friends and have this fixed everywhere.

Comments welcome.

JohnVidler, 2024-04-15:
Created a PR for this over in https://github.com/lancaster-university/codal-core/pull/169 - which adds better version symbol handling.

microbit-carlos, 2024-07-10:
These two PRs has been merged and this issue can be closed as "completed":
- https://github.com/lancaster-university/codal-core/pull/169
- https://github.com/lancaster-university/codal-microbit-v2/pull/434

These get the provided defines and function outputs:

```
CODAL_VERSION_MAJOR: 0
CODAL_VERSION_MINOR: 2
CODAL_VERSION_PATCH: 66
CODAL_VERSION_HASH: 8c8db00
CODAL_TARGET_NAME: codal-microbit-v2
CODAL_VERSION: 0.2.66
CODAL_VERSION_LONG: 0.2.66+codal-microbit-v2-8c8db00
DEVICE_DAL_VERSION: 0.2.66
MICROBIT_DAL_VERSION: 0.2.66
microbit_dal_version(): 0.2.66
uBit.getVersion(): 0.2.66
```

closed 2024-07-10 by microbit-carlos