lancaster-university/codal-microbit-v2 #146: datalog: allow online HTML file to alert users of firmware updates

jaustin, 2021-09-13:
@microbit-mark I'd like a proposal from you on how we display this, but once #145 is complete, @microbit-matt-hillsdon will be able to add DAPLink update notification to the online HTML file in the datalog.

This ticket is to track the implementation of the message in the online HTML, as opposed to adding (or @finneyj showing us how to extract) the information into the offline HTML so the online version can respond to it.

@microbit-mark could you please propose some language and styling to the online HTML experience

CC @microbit-giles

microbit-carlos, 2022-05-13:
> @microbit-mark could you please propose some language and styling to the online HTML experience

Hi Mark, would you be able to look into this next week? John will be looking into this soon.

JohnVidler, 2022-05-13:
Related to #145

microbit-mark, 2022-05-25:
A couple of suggestions. In both cases, I'm assuming that we are linking to https://microbit.org/get-started/user-guide/firmware/ unless we have a good way of downloading the exact firmware they need.

Even if we are able to provide the correct firmware .hex, we may need to still guide them in updating it

[image]

[image]

microbit-carlos, 2022-06-10:
Moving this to the next milestone as the tasks for 0.2.39 is https://github.com/lancaster-university/codal-microbit-v2/issues/145, we can update the online JS at a later point.