lancaster-university/codal-microbit-v2 #150: if a uBit.sleep() is used followed by target_wait_us() the delays do not match

microbit-carlos, 2021-11-05:
I expect the 4ms resolution of `uBit.sleep()` as that is the ticker configuration, however it'd expect `target_wait_us()` to not have this limitation as it spin waits using the timer peripheral.

To test via pin toggle with a logic analyser we can first confirm that `target_wait_us()` works fine on its own:

```cpp
while (true) {
    uBit.io.P15.setDigitalValue(1);
    target_wait_us(4000);
    uBit.io.P15.setDigitalValue(0);
    target_wait_us(5000);
    uBit.io.P15.setDigitalValue(1);
    target_wait_us(6000);
    uBit.io.P15.setDigitalValue(0);
    target_wait_us(7000);
    uBit.io.P15.setDigitalValue(1);
    target_wait_us(8000);
    uBit.io.P15.setDigitalValue(0);
    target_wait_us(9000);
}
```

Ignoring the inaccuracy of `target_wait_us` (4ms delay is measured as 3.85ms), we can clearly see the delay increasing proportionaly:
[image]

Adding a `uBit.sleep(4);` (minimum delay) with each `target_wait_us()` and the results change:
```cpp
while (true) {
    uBit.io.P15.setDigitalValue(1);
    // This delay is 7.85ms -> looks "fine", similar inaccuracy as the previous test
    uBit.sleep(4);
    target_wait_us(4000);
    uBit.io.P15.setDigitalValue(0);
    // This delay is 8.95ms -> this looks good too
    uBit.sleep(4);
    target_wait_us(5000);
    uBit.io.P15.setDigitalValue(1);
    // This delay is ~13ms ❌
    uBit.sleep(4);
    target_wait_us(6000);
    uBit.io.P15.setDigitalValue(0);
    // This delay is ~13ms ❌
    uBit.sleep(4);
    target_wait_us(7000);
    uBit.io.P15.setDigitalValue(1);
    // This delay is  ~13ms ❌
    uBit.sleep(4);
    target_wait_us(8000);
    uBit.io.P15.setDigitalValue(0);
    // This delay is  ~13 -> I guess this is finally correct
    uBit.sleep(4);
    target_wait_us(9000);
    uBit.io.P15.setDigitalValue(1);
    // This delay is ~17ms ❌
    uBit.sleep(4);
    target_wait_us(10000);

    // Stay low for 40ms to easily find the start of the next iteration
    uBit.io.P15.setDigitalValue(0);
    uBit.sleep(40);
}
```

[image]

martinwork, 2021-11-07:
My understanding is that, if uBit.sleep(4) is called just after a tick, the 4ms will elapse just after the next tick, so the sleep won't be ended until at least the tick after that. If there are other fibers, uBit.sleep won't return until they all block. That is, uBit.sleep(4) means sleep for at least 4ms.

microbit-carlos, 2021-11-07:
Right, that makes sense, thanks!
So `uBit.sleep(4)` can last from 4 to 8 ms, and that would explain what we see in these measurements.

martinwork, 2021-11-08:
Yes, close to 4ms if you're lucky, and likely noticeably over 8ms if the ticker is held up by higher priority interrupts, or there are other fibers that don't yield (e.g. serial.send(SPINWAIT)). The ticker just moves sleeping fibers who's time is up from the asleep list to the end of the run queue of fibers to be executed one at a time, round-robin style, as each sleeps or waits on an event.

microbit-carlos, 2022-05-05:
While technically this is not a bug it would be good to document this behaviour.

microbit-carlos, 2023-01-18:
I'll close this now as there is no need to keep this issue open, the additional docs will be added in this PR:
- https://github.com/lancaster-university/codal-microbit-v2/pull/248

closed 2023-01-18 by microbit-carlos