lancaster-university/codal-microbit-v2 #151: Time format is confusing in visual preview

microbit-mark, 2021-11-16:
The mixed-use of commas and periods in the time stamp on the visual preview tripped several people up in our recent data logging user study. It looks like there are 3 comma-separated values here where infact it is just time and sound value

[image]

The time appears without the additional comma in the table view

[image]

Can we consolidate on one format for representing time?

microbit-matt-hillsdon, 2021-11-16:
This is the Plotly default tooltip for a cartesian point. It wasn't immediately clear from their docs whether we can change this, though the library is generally very configurable. Some investigation required once we're clear on what we want to do.

JohnVidler, 2022-05-13:
Just a note here to say that this is also locale-dependant, so some will show stops and commas inverted too, just to add to the chaos :)