lancaster-university/codal-microbit-v2 #152: uBit.flash.read() doesn't break down transmissions to 1KB of data

microbit-carlos, 2021-11-22:
In the spec we indicate the interface will have a 1KB buffer size for data, so we should ensure `uBit.flash.read()` only sends a max of 1KB of data per I2C transaction.

`uBit.flash.write()` breaks down the data in 64 byte blocks due to a different issue, the same can be done for `uBit.flash.read()` with 1KB blocks.

JohnVidler, 2022-06-15:
I'm going to bump this to *41, as right now, the implementation works, and its too close to a release to test any modifications with rigor, if it proves to be an issue for users, we could look to perhaps including a patch sooner rather than later

JohnVidler, 2022-08-12:
Bumping again, as this is still stable right now, if not ideal, and the changeset for 0.2.41 is getting mighty long