lancaster-university/codal-microbit-v2 #162: Bluetooth Custom Advertising Configuration

bsiever, 2022-01-24:
# Description of problem

Some applications benefit from customizing the BLE advertising data (to allow advertising custom services, advertising appearance, etc.).  The current implementation of `MicroBitBLEManager` doesn't provide any ability for extensions to change advertising.

# Possible solution

It may be possible to change the declaration and scope of [`microbit_ble_configureAdvertising( bool connectable, bool discoverable, bool whitelist, uint16_t interval_ms, int timeout_seconds, ble_advdata_t *p_advdata)`](https://github.com/lancaster-university/codal-microbit-v2/blob/8d17105316628b0d69f0d0b123d13c5f13f7743a/source/bluetooth/MicroBitBLEManager.cpp#L1174) to allow it to be customized by:

* Remove the `static` scope. 
* Add a compiler attribute for a weak reference to a default implementation, like gcc's [`__attribute__((weak))`](https://gcc.gnu.org/onlinedocs/gcc-4.3.5/gcc/Function-Attributes.html#:~:text=on%20line%205.-,weak,a.out%20targets%20when%20using%20the%20GNU%20assembler%20and%20linker.,-weakref).  

This would allow extension writers to provide a substitute implementation if desired. 

## Alternative 

A callback could also be included to allow user updates to advertising data structures before they are delegated to the BLE stack. I.e., `MicroBitBLEManager` would support some mechanism to specify a callback that would be called when configuring advertising data.  That callback would update data as needed for the advertising customization. It could be called with all relevant data structures before [`ble_advdata_encode` and `sd_ble_gap_adv_set_configure`](https://github.com/lancaster-university/codal-microbit-v2/blob/8d17105316628b0d69f0d0b123d13c5f13f7743a/source/bluetooth/MicroBitBLEManager.cpp#L1199) were called.

# Concerns 

The major concerns with the primary approach described:

1. Does the toolchain being used support something comparable to `__attribute__((weak))`?
1. The full set of responsibilities of the replaceable `microbit_ble_configureAdvertising` would need to be well defined to ensure user code didn't omit anything. 
1.  It doesn't appear that this pattern (weak function attributes) are used in any other aspect of CODAL.  It would be an unusual outlier and not readily apparent to developers. 
1. It would add another global symbol. 

The alternative proposed requires additional review / consideration to be sure it provides enough flexibility.

microbit-carlos, 2022-05-23:
@bsiever we'd be curios to hear about some of the specific uses cases for this kind of change?

bsiever, 2022-05-23:
@microbit-carlos Part of the intent of the ad is allowing other devices to find a service of interest...so the main use-cases are for BLE services that are discovered via advertisements.

I needed this for a MakeCode HID extension (keyboard/mouse/etc): https://github.com/bsiever/microbit-pxt-blehid .  If a device doesn't advertise as a HID device, the host OS (Windows, macOS, Android, iOS) won't list it in the discovered devices.  It would be helpful for many other BLE profiles, like heart rate, etc. 

A workaround _is_ possible, but inelegant.  I had to duplicate a lot of the existing CODAL advertisement setup and inject the few changes I needed (appearance & UUID): [here](https://github.com/bsiever/microbit-pxt-blehid/blob/2783a418e6a67c2d26eef24e8e2be2ad0c64e5d1/HIDService.cpp#L335).    I worry that my workaround may be undermined by future changes to CODAL.

If there are not many other use-cases that are of interest, I understand if this won't be pursued. 

Happy to discuss it more if needed!