lancaster-university/codal-microbit-v2 #163: Datalog: Enabling debug breaks the long reset button press

microbit-carlos, 2022-01-25:
With a new clone of microbit-v2-samples:

```cpp
#include "MicroBit.h"

MicroBit uBit;

int  main() {
    uint32_t show = 0;
    uBit.init();
    uBit.log.setTimeStamp(TimeStampFormat::Milliseconds);

    while (true) {
        uBit.display.printChar(show++ % 2 ? '.': ' ');
        uBit.sleep(20);
    }
}
```

This works fine in a micro:bit V2.0 and V2.2 with their factory DAPLink images. The target blinks an LED in the matrix display, then a reset button long pressed (keeping it pressed down until the red LED is turned OFF) puts the board in "stand-by" mode (USB connected, red LED blinking, target MCU in sleep).

But if the `"DMESG_SERIAL_DEBUG": 1` option is added to enable debug messages, the long reset button press doesn't work anymore. After releasing the button the red LED is stable ON and the target continues running the programme and blinking an LED in the display matrix:

```json
{
    "target": {
        "name": "codal-microbit-v2",
        "url": "https://github.com/lancaster-university/codal-microbit-v2",
        "branch": "master",
        "type": "git",
        "test_ignore": true,
        "dev": true
    } ,
    "config":{
        "NO_BLE": 1,
        "MICROBIT_BLE_ENABLED" : 0,
        "MICROBIT_BLE_PAIRING_MODE": 0,
        "DMESG_SERIAL_DEBUG": 1
    }
}
```

martinwork, 2022-01-26:
@microbit-carlos I suspect that transferring "relocate vtor to 0x0 -> 0x20002200 0x20002200" to serial takes up enough time, that the sleep(10) in MicroBit::init() is not long enough for MicroBitPowerManager::idleCallback() to get called 30 times, then uBit.log.setTimeStamp overwrites the shutdown request, as would any other I2C interface request.

microbit-carlos, 2022-01-26:
Good thinking!
Adding a `uBit.sleep(30);` between `init()` and `log.setTimeStamp()` does seem to help (at least on V2.20 with factory DAPLink).

microbit-carlos, 2022-05-13:
I'd be good to check if a C++ progamme without debug can trigger this.

If something like this works fine, then I'd leave this to the bottom of the backlog:

```
void main() {
  uBit.init();
  // Some I2c transaction here, maybe a datalogging call
}
```