lancaster-university/codal-microbit-v2 #169: Sleep by long press on reset button doesn't work when BLE is enabled

martinwork, 2022-02-16:
A long press on the reset button should put micro:bit into sleep mode, but this doesn't work when BLE is enabled - even just pairing mode.

AIUI, PowerManager should receive the shutdown request in https://github.com/lancaster-university/codal-microbit-v2/blob/daplink-interface-versioning/source/MicroBitPowerManager.cpp#L341 when it's idleCallback has been called MICROBIT_USB_INTERFACE_IRQ_THRESHOLD (=30) times, usually inside this sleep https://github.com/lancaster-university/codal-microbit-v2/blob/daplink-interface-versioning/model/MicroBit.cpp#L267, but possibly in a later sleep, for example if DMESG is enabled.

But a call to DAPLink before sufficient sleep, like power.getVersion() in https://github.com/lancaster-university/codal-microbit-v2/blob/daplink-interface-versioning/model/MicroBit.cpp#L217 causes the shutdown request to be lost.

The version is read to add it to the device information service information.

Could the PowerManager shutdown check be done directly, rather than inside a sleep()?
Does it need to wait for IRQ to be held active for 30 checks on start up?

martinwork, 2022-02-16:
https://github.com/lancaster-university/codal-microbit-v2/pull/171 seems to fix the current problem, but I'm not sure if it will be OK with future uses of these messages. Only MICROBIT_UIPM_EVENT_RESET_LONG_PRESS is currently used. I suppose the message could be read and stored early on, then actioned later.

microbit-carlos, 2022-02-23:
This has been resolved via https://github.com/lancaster-university/codal-microbit-v2/pull/171, thanks Martin! 🎉