lancaster-university/codal-microbit-v2 #172: UART on recieve

RTGRDH, 2022-02-18:
Hi,
I need to apply Cristian´s Algorithm to syncronize the runtime between the micro:bit V2 and a mobile application that I'm developing.

I've done a lot of testing and it seems like the the function "bluetooth.onUartDataReceived()" is at fault.

I am only able to access this method 61 times over Bluetooth UART Service, not more than that. The micro:bit does not print an error message. After 61 accesses the micro:bit does not perform anything.

I'll attach a screenshot of the micro:bit code.
[image]

Thanks in advance.

martinwork, 2022-02-18:
61 is the size of the BLE UART buffer in MakeCode.
https://github.com/microsoft/pxt-microbit/blob/d80514bda223f729d452c1e51047f3272a2317fb/libs/bluetooth/bluetooth.cpp#L95

Maybe the hold up is because the buffer is full.
Does it work when `onUartDataReceived` reads the received data?

`data = bluetooth.uartReadUntil(serial.delimiters(Delimiters.Hash))
`

RTGRDH, 2022-02-18:
Well that explains a lot.
Do I need to save the data in a variable some how?
All I want to do is to send a trigger message to the micro:bit telling it to reply via UART with a runtime.
Is there any way to clear the buffer, or is the only way to clear it by storing the received data into a variable?

martinwork, 2022-02-18:
`uartReadUntil` is the only MakeCode method I can see to pull data out of the buffer, so reading it into a variable seems like the only way. If you're only sending hash, I guess the received data will be empty, but reading it should pull the hash out of the buffer.

martinwork, 2022-02-19:
@RTGRDH That's an interesting question, but I'm afraid I don't know how this is intended to work.

It feels like it should work like that, but apparently it doesn't. It doesn't seem to work with the equivalent serial functions either. Looking at the code, it doesn't look like it would work like that.

There is onUartDataReceived("#:",.. and uartReadUntil("#:"..., and I have now found readBuffer, but I'm not sure that would help.

I would use a single onUartDataReceived with a single delimeter and use the data from uartReadUntil to determine the purpose.

RTGRDH, 2022-02-22:
I did as you described with a single delimiter and sending additional chars to determine the purpose. 
Thanks a lot.

closed 2022-02-22 by RTGRDH