lancaster-university/codal-microbit-v2 #179: Redirecting serial pins does not set the RX pin idle state to high

microbit-carlos, 2022-04-01:
Configuring the serial pins to the edge connecto pins 0 (TX) and 1 (RX) does not set pin1 idle state to high.

Taking in consideration there is an external pull up in that pin, then it's likely that the internal pull has been configure to pull down?

Simple example code & hex: [MICROBIT.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/8399416/MICROBIT.hex.zip)

```cpp
#include "MicroBit.h"

MicroBit uBit;

int main() {
    uBit.init();
    uBit.serial.redirect(uBit.io.P0, uBit.io.P1);

    while (1) {
        uBit.display.scroll("uart");
        // uBit.serial.send("ABC");
    }
}
```

In this case we a voltimeter (or scope, or anything else) we can see that pin 0 idle state is high, and pin 1 is set low.

On a V1 with DAL and both pins are set high.

martinwork, 2022-04-01:
Adding info...

V1 pulls are explicitly configured PullUp by mbed in https://github.com/lancaster-university/mbed-classic/blob/master/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/serial_api.c#L87

nrfx_uarte only ever explicitly sets NOPULL, and I don't see anything about configuring pull here: https://infocenter.nordicsemi.com/index.jsp?topic=%2Fps_nrf52833%2Fuarte.html&cp=4_1_0_5_31_7&anchor=concept_wmv_f2m_wr
But it does say "PSEL.RXD, PSEL.RTS, PSEL.RTS, and PSEL.TXD must only be configured when the UARTE is disabled."

NRF52Serial appears not to set the pins' pull mode, or detach them from previous assignments, and does change the pins when UARTE is enabled. https://github.com/lancaster-university/codal-nrf52/blob/master/source/NRF52Serial.cpp#L212

microbit-carlos, 2022-04-04:
Thanks Martin!

Also, I haven't read the UART standard (I'm not even sure if there is an official standard), but it's possible that the idle state pull up is meant to be provided by the "sender", so maybe the micro:bit is only meant to pull up the TX signal. If that's the case, then technically the V1 DAL could be considered incorrect here, although we've never encountered an issue due to this.
But once again, this is just speculation, we might have to check any kind of UART standard that might exists (maybe the Nordic datasheet could contain some info?).

jaustin, 2022-06-23:
@JohnVidler proposes that this should be pulled high when idle, according to other serial standards, so this should be fixed

JohnVidler, 2022-08-12:
PR'd over at https://github.com/lancaster-university/codal-core/pull/150 and merged into 0.2.41 for testing in BETA, closing this for now, but happy to reopen this if the changes in codal-core PR 150 cause any unexpected side effects.

closed 2022-08-12 by JohnVidler