lancaster-university/codal-microbit-v2 #181: Sound emoji logarithmic interpolation does not pass end frequency

riknoll, 2022-04-12:
The sound expression code is not passing the end frequency for logarithmic interpolation like it does with the other interpolation functions:

https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/SoundExpressions.cpp#L226

However, the implementation is expecting it:

https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/SoundSynthesizerEffects.cpp#L102

As a result sounds that use logarithmic interpolation either play as clicks or don't play at all.

riknoll, 2022-04-12:
@finneyj FYI

JohnVidler, 2022-06-08:
May be related to https://github.com/microsoft/pxt-microbit/issues/4720 - possibly not dealing with never reaching the end frequency?

JohnVidler, 2022-06-08:
Added an end frequency parameter as per the other interpolation options, see https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/SoundExpressions.cpp#L228

JohnVidler, 2022-06-08:
See https://github.com/microsoft/pxt-microbit/issues/4720#issuecomment-1150483219 for more details, but I'm going to call this done for now.

closed 2022-06-08 by JohnVidler