lancaster-university/codal-microbit-v2 #183: sound level reading is different on battery and on USB

jaustin, 2022-05-04:
(I was sure that this issue was filed, but can't see it anywhere except https://github.com/microsoft/pxt-microbit/issues/4296 tangentially)

If you make this project and use it on battery (especially older or cheap batteries) https://microbit.org/projects/make-it-code-it/disco-lights/ then the script doesn't run well (the display is much brighter on battery).

This seems to suggest the microphone absolute values are different on USB and battery, and perhaps that the recording will be impacted.

JohnVidler, 2022-05-04:
So I've done a quick test of this directly in CODAL using the uBit.audio.level object, and it seems to work fine... buuuut when it's on battery the maximum brightness for the LEDs is lower (3v0 vs 5v0) so the overall effect is a little diminished.

[image]
(On battery, brightness set to a fixed 255, showing a dimmed screen)

[image]
(On USB, brightness set to a fixed 255, showing the screen brighter)

The effect is more pronounced in person, as the camera rather boosts it a bit.

But as we use an NCP114BMX330TCG for our LDO, (not a boost converter) we're just going to get a flat `3v0 - diode losses` from the battery, or whatever it can manage, whereas we'll get a nice actual 3v0 from USB as the LDO had the headroom.

martinwork, 2022-05-04:
These graphs show raw microphone values (labelled sound) and calculated sound levels, starting with both batteries (not brand new) and USB power supply (I get the similar values with PC USB power). The sound was a constant middle C sine wave through a headphone resting on micro:bit.

After a few seconds the USB power was switched off, then on again, etc..

I guess the increased amplitude of the raw sound variations on battery-only explain the larger calculated sound level values (a similar calculation to CODAL's), though the difference is small.

[image]

A similar recording of sound level using MakeCode and spreadsheet:

[image]

JohnVidler, 2022-06-08:
Ok, so I think what we're seeing here is that while there is a _slight_ (approx. 2% ?) drop on the microphone values, which as we're not measuring a precise dB value seems acceptable, and this is compounded by the LEDs being dimmer on lower voltage source effectively scaling the output when users just taking the mic value and setting the brightness.

I can put a board on my power analyser and see if the LED dimming is just a current sourcing issue (I strongly suspect it is), but as this isn't something we can fix, I suggest we mark this as a WontFix and close, and instead use this as a teaching point for on the https://microbit.org/projects/make-it-code-it/disco-lights/ tutorial. Explain that the current the batteries can supply is lower than via USB, and maybe pose the option of using a multiplier on the mic value as an intro to scaling the sensor values?

Thoughts?

closed 2022-08-11 by JohnVidler