lancaster-university/codal-microbit-v2 #184: Remove unused motion sensor drive

microbit-carlos, 2022-05-04:
the FXOS driver is not used, so we can remove it and the auto-detection with the heap allocation can be removed as well.

microbit-carlos, 2022-06-08:
@JohnVidler The autodetect functions can be found in this repo:
- https://github.com/lancaster-university/codal-microbit-v2/blob/v0.2.38/source/MicroBitAccelerometer.cpp#L54
- https://github.com/lancaster-university/codal-microbit-v2/blob/v0.2.38/source/MicroBitCompass.cpp#L56

And the drivers itself is in the codal-core repo:
- https://github.com/lancaster-university/codal-core/blob/master/source/drivers/FXOS8700.cpp

JohnVidler, 2022-06-08:
@microbit-carlos Just checking that these are all OK to just outright remove, and that they have no dependencies?

Would it be prudent to include stubs with `__attribute__ ((deprecated))` tagged on to indicate why this will not work going forwards?

JohnVidler, 2022-06-09:
So, I've had time to look at this now, and do we still need to have a fallback software Accelerometer class to handle cases where the LSM303 fails? if so, this boils down to simply removing this block of code ( https://github.com/lancaster-university/codal-microbit-v2/blob/v0.2.38/source/MicroBitAccelerometer.cpp#L77 ):

```
        // All known accelerometer/magnetometer peripherals have the same alignment
        if (FXOS8700::isDetected(i2c, 0x3E))
        {
            FXOS8700 *fxos = new FXOS8700(i2c, irq1, coordinateSpaceFXOS8700, 0x3E);
            MicroBitAccelerometer::detectedAccelerometer = fxos;
            MicroBitCompass::detectedCompass = fxos;
        }
```

With no further changes needed. Essentially we then just probe for an LM303, if one responds, use it, else fall back to the software one to prevent hard crashes and/or lockups in cases where it may have failed.

Thoughts?

JohnVidler, 2022-06-09:
Ok, so I've done more than the above, and greatly minimised the interface internals, removed the old driver, and added in some failure cases for if we can't configure the accelerometer and/or compass on startup.

Ultimately this will be squashed down further for `v0.2.40` or future releases, but this is acceptably squashed for now.

closed 2022-06-09 by JohnVidler