lancaster-university/codal-microbit-v2 #189: `uBit.audio.soundExpressions.play()` (blocking version) continues to output sound after the function returns

microbit-carlos, 2022-06-07:
With this programme (there is more than the screenshot), which toggles a pin high while the `[play sound until done]` block is running: [microbit-lonely-duckling-debug.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/8855622/microbit-lonely-duckling-debug.hex.zip)

[image]

We can see the sound output is actually delayed:

[image]

That `[play sound]` block in MakeCode it ultimately ends executing `uBit.audio.soundExpressions.play()` in: 
- https://github.com/microsoft/pxt-microbit/blob/d80514bda223f729d452c1e51047f3272a2317fb/libs/core/soundexpressions.cpp#L11

JohnVidler, 2022-07-12:
Proposed fix is in PR #210 and if approved will auto-close this issue as solved.

microbit-carlos, 2022-08-18:
Reopening as this might need further work: https://github.com/lancaster-university/codal-microbit-v2/pull/210#discussion_r948913835

microbit-carlos, 2022-09-20:
The original patch has also been reverted in v0.2.42-hotfix.1: https://github.com/lancaster-university/codal-microbit-v2/commit/e8e42b2ad8c0218b17ea2cbb9ef147ff94afa062
And in master:  https://github.com/lancaster-university/codal-microbit-v2/commit/696db50150fcde7f97e41fc41efe5159f19c68a7

finneyj, 2023-03-08:
Resolved, as discussed in: https://github.com/lancaster-university/codal-microbit-v2/issues/231

closed 2023-03-17 by microbit-carlos