lancaster-university/codal-microbit-v2 #195: datalog HTML: Visual Preview error for non-sequential time stamps might be confusing

microbit-carlos, 2022-06-14:
It can be fairly common for a datalog to have non-sequential time stamps, for example when a programme is logging data from start up and the reset button is pressed, then the data continues being logged but time stamp has been reset.

In this cases, when opening the "Visual Preview" is are presented with this:

[image]

A few things about the error message:

> The graph only includes rows down to row 4

1) I originally interpreted this as being "including row 4". I believe the wording here is tricky, as right now it is correct, and we want to keep the mention to the first erroneous row, as it actually has a link to it.
So changes like using `until row 3` are not valid.
Not sure I have a better suggestion at the moment.

2) Should this be row 4 or row 3?
Maybe this will depend on the person, but I guess I would consider this to be a table with a header and 3 rows, instead of a 4 rows table.
For example, if I were to ask somebody to tell me what's the first row, I'd expect them to say `1, 1` instead of `Time, a`.
  

> The visual preview does not support repeated headers or non-numeric data. The first column must be a timestamp.

3) These are valid error conditions, but none of these apply in this use-case. We should also add non-sequential time stamps to this message.

microbit-carlos, 2022-06-14:
Oh, and in case it's useful this MakeCode programme was used to generate the reproducer:

```js
datalogger.includeTimestamp(FlashLogTimeStampFormat.None)
datalogger.log(
datalogger.createCV("Time (seconds)", 1),
datalogger.createCV("a", "1")
)
datalogger.log(
datalogger.createCV("Time (seconds)", 2),
datalogger.createCV("a", "2")
)
datalogger.log(
datalogger.createCV("Time (seconds)", 1),
datalogger.createCV("a", "3")
)
```

martinwork, 2024-05-03:
Reported in support ticket https://support.microbit.org/helpdesk/tickets/74936 (private)

martinwork, 2024-05-20:
A comment was added to the above ticket that very often RESET gets pressed, on purpose or by accident, with an additional suggestion that it would e useful to read the last values from the logged data into variables.

I wonder if it would be possible to fall back to using row number for the horizontal axis.

martinwork, 2024-07-26:
Reported in support ticket https://support.microbit.org/helpdesk/tickets/80176 (private)