lancaster-university/codal-microbit-v2 #203: MakeCode power extension programme does not go to sleep

microbit-carlos, 2022-06-27:
## Description

This programme does seem to go to sleep.
Although it's possible it goes to sleep but something wakes it up immediately.

Changing the `[request low power]` for a `[request low power for [10000] ms` block and it seems to work.

## Test Programme

[image]

```ts
input.onButtonPressed(Button.A, function () {
    basic.showString("A")
    power.lowPowerRequest()
})
basic.forever(function () {
    basic.showNumber(0)
    basic.showIcon(IconNames.Heart)
})
```

## Hardware & Software

Tested on a V2.2 using the micro:bit power management extension and MakeCode live.

MakeCode info:
- makecode.microbit.org version:  [5.0.6](https://github.com/Microsoft/pxt-microbit/releases/tag/v5.0.6)
- Microsoft MakeCode version:  [8.0.3](https://github.com/Microsoft/pxt/releases/tag/v8.0.3)
- microbit runtime version:  [v2.2.0-rc6](https://github.com/lancaster-university/microbit/releases/tag/v2.2.0-rc6)
- codal-microbit-v2 runtime version:  [v0.2.40](https://github.com/lancaster-university/codal-microbit-v2/releases/tag/v0.2.40)

Extension:
- https://github.com/microbit-foundation/pxt-microbit-v2-power
- v0.2.3

Share link and hex:
- https://makecode.microbit.org/_eEL9T4CEear
- [microbit-Pwr-test-sleep.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/8994684/microbit-Pwr-test-sleep.hex.zip)
m

JohnVidler, 2022-07-04:
Placing this in the v0.2.41 release for now, mainly to facilitate tracking it

martinwork, 2022-08-13:
The power manager blocks low power mode if there are no wake up sources or if a timed wake up is due to trigger soon. A button handler does not automatically create a wake up source. See https://makecode.microbit.org/_Xwxhku8zqVUm. The forever loop has a pause because of https://github.com/lancaster-university/codal-microbit-v2/issues/201#issuecomment-1214165440
[image]

microbit-carlos, 2022-08-17:
Thanks for the clarification Martin!
So by adding the `[full power on [button A]]` block it then goes to sleep.

Was this done for a specific technical reason? In my opinion it should still go to sleep even if a wake up source wasn't configured, but there might be a reason why that's not preferable?

martinwork, 2022-08-17:
> In my opinion it should still go to sleep even if a wake up source wasn't configured

I guess we didn't think of a use for that. This controls whether sleep happens.
https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/MicroBitPowerManager.cpp#L708

It looks like the only technical consideration is if a time based wake up is about to trigger. In fact, I think it will always wake up when the USB is connected, and we didn't think we needed a control for that.

JohnVidler, 2022-09-12:
I _think_ this behaviour is now as expected as of 0.2.42, but I'm going to leave this open until I've confirmed this with a beta MakeCode build.

microbit-carlos, 2022-09-12:
MakeCode test build with CODAL 0.2.42:
- https://makecode.microbit.org/app/7a9d7cf0337ed8d53fc507a899a5d414ee59a54d-9c844136fb
- https://github.com/microsoft/pxt-microbit/pull/4829

microbit-carlos, 2022-09-15:
> I _think_ this behaviour is now as expected as of 0.2.42, but I'm going to leave this open until I've confirmed this with a beta MakeCode build.

I think the behaviour described in the original comments (not going to sleep unless a wake up source is configured) was by design in the extension? (which we might want to change)

Has something changed in CODAL to change this?

microbit-carlos, 2022-09-20:
Testing this with the latest CODAL and it doesn't look like CODAL is going to sleep, so not something within the extension, but something to change in CODAL:

```cpp
#include "MicroBit.h"

MicroBit uBit;

int main() {
    uBit.init();

    while (true) {
        uBit.power.deepSleep();
        uBit.display.print("A");
        uBit.sleep(500);
    }
}
```
[MICROBIT.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/9607923/MICROBIT.hex.zip)

martinwork, 2022-09-20:
See
https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/MicroBitPowerManager.cpp#L740

It's probably returning here
https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/MicroBitPowerManager.cpp#L773

JohnVidler, 2022-09-20:
Done. Essentially removed the checks for going to a 'blind' deep sleep, and allow the board to do it. Furthermore, its now possible to wake the board with serial data from USB by default, so there should always be a wake source of some kind by default, even if the user never configures one (unless they manually disable wake-on-USB, of course)

JohnVidler, 2022-09-21:
This is all now merged into master, pending being tagged - but I'll leave this open until we've built againts PXT and confirmed that this is solved... but I think this is now solved :)

microbit-carlos, 2022-09-22:
Test URL for v0.2.43: https://makecode.microbit.org/app/37f6efa929bcadb89d3f5b689c0cf3cae27e49d1-df227f1388

microbit-carlos, 2023-03-02:
@JohnVidler can you test this with the latest CODAL pxt test deployment?

microbit-carlos, 2023-07-25:
This working in the latest CODAL/MakeCode.