lancaster-university/codal-microbit-v2 #204: Datalog: Simple hello world CODAL programme doesn't erase data in datalog

microbit-carlos, 2022-07-01:
First flash this MicroPython datalogging programme, press the buttons a couple of times and confirm there is data in the log

Using v0.2.40.
```cpp
#include "MicroBit.h"

MicroBit uBit;

int  main() {
    uBit.init();

    while (true) {
        uBit.display.scroll("HELLO WORLD!");
    }
}
```

JohnVidler, 2022-07-13:
@microbit-carlos It might just be (from looking through the erase logic) that we just need `CONFIG_MICROBIT_ERASE_USER_DATA_ON_REFLASH` by default in `codal.json` for new projects... I'll test this and if its the case we can perhaps just bump this flag into the default `codal.json` config.

martinwork, 2022-07-14:
@JohnVidler @microbit-carlos
CONFIG_MICROBIT_ERASE_USER_DATA_ON_REFLASH is defaulted to 1 here
https://github.com/lancaster-university/codal-microbit-v2/blob/13a651754e668464855fc40a81dd7dcb796b58c9/model/MicroBit.h#L107

The problem is that, when MicroBitLog::invalidate() is called, MicroBitLog::_IsPresent() is returning false here
https://github.com/lancaster-university/codal-microbit-v2/blob/897e18b11fdfbd00123db0b435b949e7ad95e8e5/source/MicroBitLog.cpp#L973
because the python program is storing
MICROBIT_LOG_VERSION = UBIT_LOG_FS_V_001
but the newly compiled CODAL program is looking for
MICROBIT_LOG_VERSION = UBIT_LOG_FS_V_002.

JohnVidler, 2022-07-29:
Not 100% sure how to proceed on this one; do we want to always reset the log _irrespective of version_, and if so, then the fix is to check via `memcmp(..., 14)` (skipping the actual version number).

Otherwise, what's the logic here? Do we only care to reset the memory if we're being reset by the same version? Or by versions equal or lower than our own (the safer option)? Or by our own version or greater (a bit risky, if the format changes)?

microbit-carlos, 2022-07-29:
> do we want to always reset the log irrespective of version

What does "always" mean in this context?
We want the log to be erased each time a hex file is flashed.

> Otherwise, what's the logic here? Do we only care to reset the memory if we're being reset by the same version? Or by versions equal or lower than our own (the safer option)? Or by our own version or greater (a bit risky, if the format changes)?

I'm not sure if the log version was accomplishing something specific, maybe an old requirement? Or maybe it was introduced before it was decided that a programme flash should always erase the log?

It could be that it's just trying to determine if there is an actual log to be erased (technically there could be anything in the DAPLink storage).

Might be worth asking Joe in case there is something not obvious.

Apart from that, when a programme is flashed, I don't think we should "not erase" the log depending on the log version, so the could reduce the check to only verify `UBIT_LOG_FS_V_` without the version number.

martinwork, 2022-07-29:
It will be possible to make new hex files ignore the version, but not old hex files.

I was aware of the version change because I had just been writing some code to interpret the MY_DATA file and extract the data. I had to ignore the version and hope. It would be good to have some way to flag that an incompatible change has been made, Maybe thresholds in the version number? Why was the version changed this time?

microbit-carlos, 2022-08-17:
> Why was the version changed this time?

The version number was increased as the DAPlink version is now included in the HTML, so in the future we can update the JS loaded from the micro:bit servers and indicate to the user if they need to upgrade their DAPlink firmware.

microbit-carlos, 2022-08-17:
I've tested this with the latest CODAL release and can confirmed it's been fixed by https://github.com/lancaster-university/codal-microbit-v2/pull/217.

closed 2022-08-17 by microbit-carlos