lancaster-university/codal-microbit-v2 #209: Errors after compile

keble6, 2022-07-09:
I installed microbit-v2-samples and tried using the python build.py command. That worked fine a few days ago on a simple Hello World..

But now I get:

codal-microbit-v2 is already installed
Set target: codal-microbit-v2 
Using target.json (dev version) 
Targeting codal-microbit-v2
Supressing -Wexpansion-to-defined.
-- Adding library path: (/home/keblerw/microbit-v2-samples-master/libraries)
Installing dependencies...
codal-core is already installed
codal-nrf52 is already installed
codal-microbit-nrf5sdk is already installed
Using library: codal-microbit-v2
Using library: codal-core
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
Using library: codal-nrf52
Using library: codal-microbit-nrf5sdk
-- Configuring done
-- Generating done
-- Build files have been written to: /home/keblerw/microbit-v2-samples-master/build
[ 33%] Built target codal-core
[ 43%] Built target codal-nrf52
[ 66%] Built target codal-microbit-nrf5sdk
[ 89%] Built target codal-microbit-v2
[ 89%] Linking CXX executable MICROBIT
/usr/lib/gcc/arm-none-eabi/8.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/MICROBIT.dir/source/main_orig.cpp.obj: in function `codal::CodalComponent::~CodalComponent()':
/home/keblerw/microbit-v2-samples-master/libraries/codal-microbit-v2/model/MicroBit.h:119: multiple definition of `main'; CMakeFiles/MICROBIT.dir/source/main.cpp.obj:/home/keblerw/microbit-v2-samples-master/source/main.cpp:5: first defined here
/usr/lib/gcc/arm-none-eabi/8.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/MICROBIT.dir/source/main_orig.cpp.obj:/home/keblerw/microbit-v2-samples-master/source/main_orig.cpp:3: multiple definition of `uBit'; CMakeFiles/MICROBIT.dir/source/main.cpp.obj:/home/keblerw/microbit-v2-samples-master/source/main.cpp:3: first defined here
/usr/lib/gcc/arm-none-eabi/8.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/MICROBIT.dir/source/myADC.cpp.obj: in function `main':
/home/keblerw/microbit-v2-samples-master/libraries/codal-microbit-v2/model/MicroBit.h:119: multiple definition of `main'; CMakeFiles/MICROBIT.dir/source/main.cpp.obj:/home/keblerw/microbit-v2-samples-master/source/main.cpp:5: first defined here
/usr/lib/gcc/arm-none-eabi/8.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/MICROBIT.dir/source/myADC.cpp.obj:/home/keblerw/microbit-v2-samples-master/source/myADC.cpp:16: multiple definition of `uBit'; CMakeFiles/MICROBIT.dir/source/main.cpp.obj:/home/keblerw/microbit-v2-samples-master/source/main.cpp:3: first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/MICROBIT.dir/build.make:444: MICROBIT] Error 1
make[1]: *** [CMakeFiles/Makefile2:239: CMakeFiles/MICROBIT.dir/all] Error 2
make: *** [Makefile:103: all] Error 2

I have updated and upgraded Linux (running under Chrome OS).
I'd appreciate any help!

martinwork, 2022-07-09:
It looks like the source folder has files main.cpp and main_orig.cpp, both of which define the main() function.
And both main.cpp and myADC.cpp define uBit. There must be only one definition of these things.

keble6, 2022-07-09:
Thanks - I have moved those other files to mysource folder:

$ ls source
main.cpp mysource samples

But I still get the problem:

/usr/lib/gcc/arm-none-eabi/8.3.1/../../../arm-none-eabi/bin/ld: 
CMakeFiles/MICROBIT.dir/source/mysource/myADC.cpp.obj: in function 
`main':
/home/keblerw/microbit-v2-samples-master/libraries/codal-microbit-v2/model/MicroBit.h:119: 
multiple definition of `main'; 
CMakeFiles/MICROBIT.dir/source/main.cpp.obj:/home/keblerw/microbit-v2-samples-master/source/main.cpp:5: 
first defined here
/usr/lib/gcc/arm-none-eabi/8.3.1/../../../arm-none-eabi/bin/ld: 
CMakeFiles/MICROBIT.dir/source/mysource/myADC.cpp.obj:/home/keblerw/microbit-v2-samples-master/source/mysource/myADC.cpp:16: 
multiple definition of `uBit'; 
CMakeFiles/MICROBIT.dir/source/main.cpp.obj:/home/keblerw/microbit-v2-samples-master/source/main.cpp:3: 
first defined here
/usr/lib/gcc/arm-none-eabi/8.3.1/../../../arm-none-eabi/bin/ld: 
CMakeFiles/MICROBIT.dir/source/mysource/orig.cpp.obj: in function 
`codal::CodalComponent::~CodalComponent()':
/home/keblerw/microbit-v2-samples-master/libraries/codal-microbit-v2/model/MicroBit.h:119: 
multiple definition of `main'; 
CMakeFiles/MICROBIT.dir/source/main.cpp.obj:/home/keblerw/microbit-v2-samples-master/source/main.cpp:5: 
first defined here
/usr/lib/gcc/arm-none-eabi/8.3.1/../../../arm-none-eabi/bin/ld: 
CMakeFiles/MICROBIT.dir/source/mysource/orig.cpp.obj:/home/keblerw/microbit-v2-samples-master/source/mysource/orig.cpp:3: 
multiple definition of `uBit'; 
CMakeFiles/MICROBIT.dir/source/main.cpp.obj:/home/keblerw/microbit-v2-samples-master/source/main.cpp:3: 
first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/MICROBIT.dir/build.make:444: MICROBIT] Error 1
make[1]: *** [CMakeFiles/Makefile2:239: CMakeFiles/MICROBIT.dir/all] 
Error 2
make: *** [Makefile:103: all] Error 2

So where do other source files have to go?

Rob

Links:
------
[1] 
https://github.com/lancaster-university/codal-microbit-v2/issues/209#issuecomment-1179587410
[2] 
https://github.com/

martinwork, 2022-07-09:
The build system picks up all the files it can compile in folder source and subfolders, so moving them into source/mysource doesn't help. Move files that aren't needed out of source. Alternatively I think you could rename main_orig.cpp to main_orig.unused, for example.

keble6, 2022-07-09:
Thanks Martin, that makes sense. If you don't hear back from me this evening then it will be solved!

Maybe a bit of help text in the readme would help?

Rob

***@***.***>

microbit-carlos, 2022-08-10:
I'll close this as resolved, but feel free to reopen if you are still experiencing issues. Thanks for the support Martin!

closed 2022-08-10 by microbit-carlos