lancaster-university/codal-microbit-v2 #211: Feature request:  ADC reference voltage change

martinwork, 2022-07-13:
Is it possible to change the ADC reference voltage with an API that can be exposed in MakeCode? 

Related:
https://github.com/lancaster-university/microbit-dal/issues/435
https://github.com/lancaster-university/microbit-dal/issues/356
https://github.com/lancaster-university/microbit-dal/issues/322

From https://support.microbit.org/a/tickets/53690 (private)

> Would it be possible to change the ADC reference voltage either permanently or by some Makecode block under user control?
> 
> This reference voltage is set to the 3V power rail of the Microbit, which means the reference voltage is unknown and variable. On battery power this reference voltage is dependant on the state of charge of the battery.
> 
> This means that measuring voltages on the analogue inputs (P0 – P2) is not accurate and therefore limits the usefulness of this for control applications and when data logging.
> 
> The ability to use a stable reference voltage does exist and the technique can be found at https://meanderingpi.com/2017/08/05/microbit-analog-inputs/ and https://gist.github.com/microbit-carlos/ with example for a V1 Microbit in Micropython.
> 
> It should be fairly easy to port this over to the Makcode editor environment and if not directly compatible to a V2, the principles should again be relatively easy to be applied.
> 
> Implementing this request will greatly enhance the usefulness of using the Microbit to measure and log accurately voltages by way of analogue read functions.
> 
>

microbit-carlos, 2023-01-17:
This feature would also useful be for Arduino API compatibility (side project, so not something that should hold any weight in the decision).

However, we might need to consider this carefully, as ADC reference voltages are specific to the underlying hardware, and CODAL aims to provide (as much as possible) a hardware agnostic generic abstraction.

lobodpav, 2025-01-05:
I'm trying to build an automatic watering solution using `micro:bit`. It will be battery-powered and use `iot:bit` from Elecfreaks to become a server (remote control) and client (data reporting), using a WiFi connection.

One key aspect is a reliable measurement of the battery level powering micro:bit and the water pump.
The first idea crossing my mind was to connect the `+` battery wire to an analogue input pin to measure the voltage.

However, without setting the internal fixed 1.2V voltage reference and the pre-scaler, it's not possible to measure the battery voltage.

This is quite an issue when building automated systems that rely on battery power. With a reliable way to measure the voltage, I could notify myself when the battery is reaching its limits so I can recharge it. Otherwise, the system will just shut down one day without any heads-up.

If I'm not mistaken, a similar issue will happen to the soil moisture sensor. With the input voltage dropping as the battery drains, the reading will get smaller values, preventing me from measuring the soil moisture reliably.