lancaster-university/codal-microbit-v2 #213: Error code 50/51 for the motion sensor should be thrown on first use

microbit-carlos, 2022-08-10:
At the moment I believe if there is an issue communicating with the motion sensor then we immediately panic with error code 050/051.

To match the V1 functionality, we should delay the panic to the first usage of the motion sensor.

This is not directly related, but in the same area as:
- https://github.com/lancaster-university/codal-microbit-v2/issues/191

martinwork, 2022-08-10:
See https://github.com/lancaster-university/codal-microbit-v2/pull/192#pullrequestreview-1003021469

JohnVidler, 2022-08-12:
This is correcly handled in `master` and will be the case for tags from `0.2.41` onwards.

The implementation allows for a weird kind of partial failure in the case of the Micro:bit v2 where our compass and accelerometer are the same physical device, and includes support for the compass (for example) failing to come up, but the accelerometer working just fine, but I've left it this way in case we have any extensions to either class in the future, or on other boards, so they can independently fail.

See also:
- https://github.com/lancaster-university/codal-microbit-v2/issues/191#issuecomment-1213239668

microbit-carlos, 2022-09-07:
As part of this ticket we should also check if there is a way to simplify a bit, or reduce how many places throw the panic, as discussed in:
- https://github.com/lancaster-university/codal-microbit-v2/pull/216#issuecomment-1216625005

closed 2023-08-30 by JohnVidler