lancaster-university/codal-microbit-v2 #218: Tracking updates with a changelog

microbit-carlos, 2022-08-18:
Ideally in a CHANGELOG file at the root of the repo, and it can optionally be added to the GitHub tags/releases as well.

If possible it'd be good to go back a few versions back, but not too far back, around the time you join @JohnVidler might be a good point.

JohnVidler, 2022-09-02:
Now addressed in a `Changelog.md` file in the samples repo - expect this in releases moving forward.

I'm going to close this as its already actioned, just not visible yet until we tag.

microbit-carlos, 2022-09-02:
In the microbit-v2-samples repo? But that one doesn't get tagged, right?

microbit-carlos, 2022-10-20:
@JohnVidler can we move the changelog to this repo and start adding entries from the v0.2.40 release? That's the MakeCode 2022 tag used.
We can check the commits between releases, but a changelog should be more "CODAL consumer focused"

microbit-carlos, 2023-03-02:
Changelog is created as part of the GitHub releases description.
The project README will be updated as part of https://github.com/lancaster-university/codal-microbit-v2/issues/281 to include this info as well.

closed 2023-03-02 by microbit-carlos