lancaster-university/codal-microbit-v2 #221: `uBit.power.deepSleep(ms)` blocks until the `ms ` time has elapsed

microbit-carlos, 2022-08-23:
Basically, it looks like if a wake up source (like a button press) is provided and `MicroBitPowerManager::deepSleep(uint32_t milliSeconds)` is used with the `milliSeconds` argument, then `deepSleep()` blocks until the delay has finished.

In this example using CODAL v0.2.40 (41 is currently broken https://github.com/lancaster-university/codal-microbit-v2/issues/220), button A puts the board to sleep for 5 seconds, and is also configured to wake up on button B.

When button A is pressed and button B pressed within the 5 seconds window, the display shows "S" until the time has elapsed, so `uBit.power.deepSleep()` is blocking.

```cpp
#include "MicroBit.h"

MicroBit uBit;

int main() {
    uBit.init();
    uBit.display.print("A");

    // Wake up via button A
    uBit.buttonA.wakeOnActive(true);

    while (true) {
        // Go to sleep on Button B
        if (uBit.buttonB.isPressed()) {
            uBit.display.print("S");
            uBit.power.deepSleep(5000);
            uBit.display.print("Up");
        }
        uBit.sleep(100);
    }
}
```

In this case it's easy to figure out what is happening because the display shows `S` and then `Up`, but I think the users would expect the programme to continue where it was left out (without blocking) as soon as it wakes up.

If the users need to block for that amount of time they can check the time before and after `uBit.power.deepSleep()` and manually block if needed.

microbit-carlos, 2022-08-23:
I'm getting errors trying to upload this file to the original message, so had to add it is as a response:
- [MICROBIT_blockingsleep.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/9404448/MICROBIT_blockingsleep.hex.zip)

martinwork, 2022-08-23:
request low power for 1000 ms / power.lowPowerPause(1000) / uBit.power.deepSleep(1000) is intended to work like basic.pause(1000) or uBit.sleep(1000). Consider this  example. https://makecode.microbit.org/_AYa35d0hWdws. I would not expect button A to rush the tones. The example I have in mind is something like:  every hour, log the average of 5 readings 10s apart, and allow button A to display the current value at any time.

If deepSleep(1000) is woken up early, it behaves like a uBit.sleep(1000), but it's wake up alarm remains in place, and it doesn't stop the fiber that wanted to wake up from requesting low power again, like the button A handler does in this example. deepSleep(1000) and uBit.sleep(1000) sleep for at least 1000ms. Perhaps we need a deepSleepForUpTo(1000).

[image]

microbit-carlos, 2022-08-30:
Thanks Martin!
That's a good point, I guess that block could be used as a "sleep/pause". In those cases I can see how users could expect the sleep time to be respected.

In the MakeCode case, the button A still reacts "on time", as the "sleep/pause" only affects the fiber where those blocks are placed. So pressing the button still wakes up the device, shows the images and goes back to sleep, and the wait times from the other fiber `[request low power]` blocks are respected.

It's a bit different in the MicroPython, specially since the user code doesn't really end up spinning fibers, except maybe for the `run_every` decorator? (@dpgeorge is that the case?). So, for MicroPython we added the wake up sources to each call to the "low power" function:
https://github.com/bbcmicrobit/micropython/pull/754/files
```python
power.deep_sleep(ms=10_000, wake_on=button_a)
```

And so, in something like this, the button press wakes up the device, but it stays there waiting before it shows the happy icon. And if there is nothing to be shown in the display at the time, then it looks like the device doesn't wake up until the time is up:

```python
display.clear()
power.deep_sleep(ms=10_000, wake_on=button_a)
display.show(Image.HAPPY)
```

If we want to keep the current MakeCode behaviour and allow MicroPython to have a non-blocking `power.deep_sleep()`, then it might need to provide an alternative version of `uBit.power.deepSleep()`, or an additional argument to select behaviour.

@JohnVidler @finneyj any thoughts and/or ideas?

dpgeorge, 2022-08-31:
> specially since the user code doesn't really end up spinning fibers, except maybe for the `run_every` decorator?

No, fibers are not used in MicroPython for anything.  `run_every` is done using a timer and callback.

martinwork, 2022-08-31:
@dpgeorge @microbit-carlos Is the scheduler running?
https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/MicroBitPowerManager.cpp#L511

If not, then maybe this override is not working in the micropython environment
https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/MicroBitPowerManager.cpp#L800

dpgeorge, 2022-08-31:
I think the scheduler is running.  How would I check?

JohnVidler, 2022-09-01:
> I think the scheduler is running. How would I check?

Technically CODAL has a `fiber_scheduler_running()` call if this is needed, but in theory it should all 'just work', as any critical sections already make this scheck for you internally.

> @JohnVidler @finneyj any thoughts and/or ideas?

Perhaps an extension to the `deepSleep` method to support a kind of 'interruptable deep sleep', which explicitly _does not_ respect the remaining time if the board is woken up prematurely? `bool interruptableDeepSleep( uint32_t milliseconds )` perhaps, with a boolean return for if the sleep was interrupted, in case this is needed later down the line for timing requirements (audio playback, display stuff, etc) ?

JohnVidler, 2022-09-01:
Possible patch (not PR'd yet) to suppor this: https://github.com/lancaster-university/codal-microbit-v2/compare/master...feature/interruptable-deep-sleep#diff-6e871d2e64de0dfdd39d7ad5a51fb9c26786e1472b8738d6a00b673db7b857d8R527

dpgeorge, 2022-09-01:
> perhaps, with a boolean return for if the sleep was interrupted,

Yes we will definitely need a return value to distinguish (eg) a button press waking it up from a timeout of the sleep.

JohnVidler, 2022-09-02:
I've now tested this with co-related patches to the FiberLock class, and all seems to be working, along with a new deepSleep() call in this form: `bool interrupted = uBit.power.deepSleep(2000, true);` which (when true in the second arg) allows for interrupted sleep events to skip any remaining wait time and just continue.

If a sleep event is interrupted, this also returns true, so that code can decide what to do - @dpgeorge / @microbit-carlos presumably this is all the functionality you'd need for Python's wake logic?

This is all part of a patchest that spans 3 repos (at the moment) with various small fixes throughout, so I'll hold off closing this for the moment while everything gets squared away, then expect quite a few issues to close all at once.

dpgeorge, 2022-09-07:
> presumably this is all the functionality you'd need for Python's wake logic?

Yes that looks good, thank you!

microbit-carlos, 2022-09-15:
This has shipped in the v0.2.42 release: https://github.com/lancaster-university/codal-microbit-v2/releases/tag/v0.2.42

closed 2022-09-15 by microbit-carlos