lancaster-university/codal-microbit-v2 #225: Sound level in MakeCode never reaches high values

microbit-carlos, 2022-08-24:
Creating this issue here first for discussion in case there was something that has changed in CODAL to cause this.

Simple MakeCode programme to print to serial the sound level:

```js
basic.forever(function () {
    serial.writeLine("" + (input.soundLevel()))
})
```

With the current MakeCode live (non-beta), using CODAL v0.2.40, it never returns a value higher that 131

microbit-carlos, 2022-08-24:
Okay, I've been going back through MakeCode version and it looks like it was never possible to get values larger than 200 😱

CODAL v0.2.37 to v0.2.40, MakeCode v4.1.41 to v5.1.1:
- Sound Level max value: **131**
- https://makecode.microbit.org/v4.1.41
- https://makecode.microbit.org/v5.1.1
- https://github.com/microsoft/pxt-microbit/blob/v4.1.41/libs/microphone/microphonehw.cpp
- [microbit-sound-v0237-v0240.zip](https://github.com/lancaster-university/codal-microbit-v2/files/9417932/microbit-sound-v0237-v0240.zip)
 

CODAL v0.2.35, MakeCode v4.1.34:
- Sound Level max value: **195**
- https://makecode.microbit.org/v4.1.34
- https://github.com/microsoft/pxt-microbit/blob/v4.1.34/libs/microphone/microphonehw.cpp
- [microbit-sound-v0235.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/9417931/microbit-sound-v0235.hex.zip)

At this point I noticed all versions behaved the same, so didn't collect additional data, but tried each MakeCode release that updated CODAL all the way back to MakeCode v3.1.21 with CODAL v0.2.12:
- Sound Level max value: **195**
- 29 Oct 2020
- https://makecode.microbit.org/v3.1.21
- https://github.com/microsoft/pxt-microbit/blob/v3.1.21/libs/microphone/microphonehw.cpp
- [microbit-sound-v0212.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/9418168/microbit-sound-v0212.hex.zip)

Older versions of MakeCode didn't compile:
- MakeCode v3.1.5 with CODAL v0.2.9 doesn't work
- MakeCode v3.1.4 with CODAL v0.2.1 doesn't work

JohnVidler, 2022-09-01:
The suggestion here is to keep the existing API, and create a new scaled API for extensions, etc. to use moving forward.

Do we want to actively deprecate the old calls for this, and/or rename them to `...dB()` calls for clarity?

microbit-carlos, 2022-09-01:
I believe we agreed to keep the old APIs as they are, and add new ones to returned a scaled value, which can then be used by MicroPython and MakeCode directly rather than scale the values on their side.

finneyj, 2022-09-08:
New APIs added as discussed. 

Default behaviour is unchanged, but there is now an optional method to used an unsigned 8 bit range rather than S.I. dB units in the SPLLevelDetector class:  https://github.com/lancaster-university/codal-core/blob/master/inc/streams/LevelDetectorSPL.h#L170

The following example code will drop the level detector into this new 8 bit mode:
```uBit.audio.levelSPL->setUnit(LEVEL_DETECTOR_SPL_8BIT);```

By default the 8 bit range is linearly mapped to the 35db...100db range, and can be configured with the two CONFIG options defined here via config.json :
https://github.com/lancaster-university/codal-core/blob/master/inc/streams/LevelDetectorSPL.h#L53
https://github.com/lancaster-university/codal-core/blob/master/inc/streams/LevelDetectorSPL.h#L58

I also took the opportunity to improving the low end the sensitivity of the level detector by adding in a low pass filter (it's now about 35db rather than 52db, which is much more useful). Finally, I also calibrated the various parameters in the pipeline to calibrate the sensor more accurately.

A good shout into the microphone can now hit a value of 255 @microbit-carlos, so should address the MakeCode bug described above.

@abchatra @dpgeorge - suggest we adopt this functionality as a common calibration for both micropython and MakeCode when convenient so that users can benefit from consistent behaviour across both languages?

abchatra, 2022-09-08:
Sounds good. @riknoll - fyi

dpgeorge, 2022-09-09:
> suggest we adopt this functionality as a common calibration for both micropython and MakeCode when convenient so that users can benefit from consistent behaviour across both languages?

Yes we will do that.

JohnVidler, 2022-10-03:
> New APIs added as discussed.

@finneyj Can we safely close this now? Has this lot been PR'd and merged?

finneyj, 2022-10-03:
Thanks @JohnVidler. Yep, all changes are now in main branch. Closing.

closed 2022-10-03 by finneyj