lancaster-university/codal-microbit-v2 #226: Method to change I2C pins

microbit-carlos, 2022-08-30:
This was possible in DAL, by accessing a protected variables from the Mbed I2C class:
- https://github.com/lancaster-university/microbit-dal/blob/602153e9199c28c08fd2561ce7b43bf64e9e7394/inc/drivers/MicroBitI2C.h#L46
- https://github.com/lancaster-university/mbed-classic/blob/master/api/I2C.h#L167
- https://github.com/bbcmicrobit/micropython/blob/master/inc/microbit/microbitdal.h#L38

Can we provide a method in CODAL to offer this feature directly to MicroPython?

finneyj, 2022-09-21:
Addressed in:

https://github.com/lancaster-university/codal-core/pull/155
https://github.com/lancaster-university/codal-nrf52/pull/42

finneyj, 2022-09-21:
Relevant methods here:

https://github.com/lancaster-university/codal-nrf52/blob/dynamic-pin-peripherals/inc/NRF52I2C.h#L70
https://github.com/lancaster-university/codal-nrf52/blob/dynamic-pin-peripherals/inc/NRF52SPI.h#L131

I assume this covers off what you need @microbit-carlos @dpgeorge ?

JohnVidler, 2022-09-23:
Bumping again, as 44 is likely to be any fixes for MicroPython, and this lot may introduce issues.

microbit-carlos, 2023-01-27:
PRs were merged in December, so assuming this was released in v0.2.45 (January) instead of v0.2.44 (November).
This is now ready for implementing in MicroPython 🎉

closed 2023-01-27 by microbit-carlos