lancaster-university/codal-microbit-v2 #228: Playing a sound effect crashes MicroPython

microbit-carlos, 2022-09-15:
With CODAL v0.2.42.

https://review-python-editor-next.microbit.org/micropython-update/

```python
from microbit import *

audio.play(audio.SoundEffect())
display.show("A")
```

The SoundEffect will play, but it will get stuck playing the last tone and play that indefinitely, never reaching the line to `display.show("A")`.

microbit-carlos, 2022-09-15:
Can't seem to be able to replicate this with a simple CODAL programme:

```cpp
#include "MicroBit.h"

MicroBit uBit;

int main() {
    uBit.init();

    uBit.audio.soundExpressions.play("happy");
    uBit.display.print("A");
    uBit.sleep(500);
    uBit.audio.soundExpressions.playAsync("happy");
    uBit.display.print("B");

    while (true) {
        uBit.sleep(2000);
        uBit.display.print("C");
    }
}
```

And the MicroPython code is not doing much else:
https://github.com/microbit-foundation/micropython-microbit-v2/blob/822b3c9e88f6f3faa4af458db7e35e1a818882de/src/codal_app/microbithal_audio.cpp#L96-L109
```cpp
void microbit_hal_audio_play_expression(const char *expr) {
    ++sound_synth_active_count;
    uBit.audio.soundExpressions.stop();

    // `expr` can be a built-in expression name, or expression data.
    // If it's expression data this method parses the data and stores
    // it in another buffer ready to play.  So `expr` does not need
    // to live for the duration of the playing.
    uBit.audio.soundExpressions.playAsync(expr);
}

void microbit_hal_audio_stop_expression(void) {
    uBit.audio.soundExpressions.stop();
}
```

microbit-carlos, 2022-09-15:
Okay, so building the same MicroPython with different CODAL versions, testing by playing `audio.play(Sound.HAPPY)` in the REPL
- 0.2.42: The sound it's played to the end (or thereabouts), but gets stuck playing the last tone, the REPL never returns
    - [MICROBIT_v0.2.42.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/9576501/MICROBIT_v0.2.42.hex.zip)
- 0.2.41: The sound is not played (just a quick click is played), and it gets stuck there, the REPL never returns
    - [MICROBIT_v0.2.41.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/9576496/MICROBIT_v0.2.41.hex.zip)
- 0.2.40: Works fine
    - [MICROBIT_v0.2.40.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/9576498/MICROBIT_v0.2.40.hex.zip)

JohnVidler, 2022-09-15:
I'm currently unable to reproduce this in CODAL, but continuing to investigage...

microbit-carlos, 2022-09-15:
Doing some "manual" git bisecting (trying a few older git commits), the issue has been introduced here: https://github.com/lancaster-university/codal-microbit-v2/commit/31095d5d9f3c1ffc524c2777173d10fe7ee9ab25

Going back to MicroPython with 0.2.42 and commenting out that line from CODAL seems to fix the issue.

JohnVidler, 2022-09-16:
I've now pushed a new tag which includes just this tiny patch; closing this for now: https://github.com/lancaster-university/codal-microbit-v2/releases/tag/v0.2.43

closed 2022-09-16 by JohnVidler

microbit-carlos, 2022-09-16:
@JohnVidler will there be a new ticket to revisit another fix instead of https://github.com/lancaster-university/codal-microbit-v2/commit/31095d5d9f3c1ffc524c2777173d10fe7ee9ab25 ? The original issue reopened? Or how will that be tracked?