lancaster-university/codal-microbit-v2 #230: Add an exposed boolean to track when audio is playing

JohnVidler, 2022-09-22:
For some applications where both audio input and output are used at the same time, it is desirable to know when audio is actually being played such that you can avoid detecting yourself with the microphone.

To facilitate this, a new API is needed which will return or access a boolean set to true whenever the synth is actively playing audio.

finneyj, 2022-09-22:
I suspect this could be achieved pretty simply in the mixer... There's already a mechanism in there to determine if the mixer is outputting silence: https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/Mixer2.cpp#L197

If we combine that with some profiling of the length of the output buffer, we should be able to create an accurate  ```bool isPlayingSound()``` method?

Alternatively, we could do something similar at the egress - and scan for empty/zero buffers reaching the NRF52PWM that drives the speaker...

microbit-carlos, 2023-01-27:
`MicroBitAudio::isPlaying()` added in https://github.com/lancaster-university/codal-microbit-v2/commit/cc02e7a54b9332cccca2a2af98eb10621c52e693.

closed 2023-01-27 by microbit-carlos