lancaster-university/codal-microbit-v2 #231: Play 'until done' should actually wait until audio is complete

JohnVidler, 2022-09-22:
Currently blocks in PXT in the form 'Play until done' do not actually wait until the audio is fully played out, leading to some overlap with the subsequent blocks.

Its suspected that this is because the underlaying CODAL calls are returning before the final buffer(s) are played out, and this needs to be investigated.

May be possible to incorporate into the same mechanisms for #230

microbit-carlos, 2022-09-22:
This is also related to:
- https://github.com/lancaster-university/codal-microbit-v2/issues/189

microbit-carlos, 2022-09-22:
I think based on the discussion in today's call we were considering having a third "mode" to the async/blocking pair:
- Async (current)
- Blocking until sent to buffer (current)
- Blocking until data is fully out (new)

finneyj, 2022-09-22:
Indeed. 

The current behaviour is there by design, as it allows back to back playouts to be possible without any glitches caused by a gap in the audio stream....

I can appreciate the need to know when the buffer has *actually* finished rendering for some applications though. I think Carlos' list is correct - there are three behaviours here, and we want them all. I'm not quite sure how that might be surfaced in blocks though.

Maybe simply:

play
play (in the background)
play (until done)

???

finneyj, 2023-03-08:
OK, so two changes now made to address this issue, both of which will be available in v0.2.50:

1) MicroBitAudio::isPlaying() has been added and merged into master branch. This will provide a very accurate indication of whether the micro:bit is producing any sound from any input to the mixer. The timings are also calibrated to account for propagation delay through the double buffered output. So this should provide a simple, fully accurate mechanism to determine if/when the micro:bit is emitting sound. This is my recommended solution to all users.

2) Against my better judgement, I've also added a new API to provide a "play until completely rendered by the speaker" option to the synthesizer as described above. It  is a relatively minor change to enable this, so I've merged it into master: 
https://github.com/lancaster-university/codal-microbit-v2/commit/18b23e44dcb3c65b191592fa9fb5fd5b31a3fc7d

Users may now do any one of:

```
uBit.audio.soundExpressions.play("happy");
uBit.audio.soundExpressions.playAsync("happy");
uBit.audio.soundExpressions.play("happy", DEVICE_SOUND_EMOJI_SYNTHESIZER_EVT_PLAYBACK_COMPLETE);
```

The change is fully backward compatible, so should have no impact on existing platforms.

HOWEVER, I do have to remind folks though that I STRONGLY advise AGAINST using this API as the default in blocks/command given to kids. if you choose to do so, this is what will happen:

- Sound will play.
- The call to play() will not exit until all the sound has completely left the speaker.
- This will create a 12ms bubble in the audio pipeline, such that (when) kids do back to back play() operations, there will be a fearsome CLICK and/or GAP before the next sound plays.
- kids and teachers get a bad experience.

My STRONG recommendation for high level languages is therefore:

```uBit.audio.soundExpressions.play("happy")``` should be the default semantic, as we currently use.
```uBit.audio.soundExpressions.playAsync("happy")```  could/should be offered as an "in the background" option where applicable
```uBit.audio.soundExpressions.play```("happy", DEVICE_SOUND_EMOJI_SYNTHESIZER_EVT_PLAYBACK_COMPLETE);``` should never be offered, or only offered with caution.

JohnVidler, 2023-05-03:
I believe that this is now resolved, but I've reassigned this to @finneyj to make the final call on that.

microbit-carlos, 2023-05-17:
Yes, based on previous conversations this is now fixed and can be closed, thanks everyone!

closed 2023-05-17 by microbit-carlos