lancaster-university/codal-microbit-v2 #233: Include build flags to remove 'softdevice' from the build

JohnVidler, 2022-09-22:
The softdevice premade binary adds signficant size to the generated builds, and is frequently not needed by users so @finneyj has constructed a branch without the softdevice additions (see linked refs), but we should really have a build switch to configure this behaviour, as removing softdevice does eliminate support for many radio functions.

This would allow high-level-languages to decide if they need this support, or would rather have the flash space, but needs significant work with CMake et.al to achieve.

NB. Putting this on v0.2.45 for now, but this is not a final decision, but just as a placeholder

JohnVidler, 2022-11-03:
Merged in to staging, and looks good so far, so on-track for the next tag.

microbit-carlos, 2022-11-08:
Are you pushing the staging branch? Can't really identify it looking at the current branches in this repo.