lancaster-university/codal-microbit-v2 #234: If MICROBIT_USB_SERIAL_WAKE is set, board wakes up from USB disconnection

microbit-carlos, 2022-09-22:
With the latest master and setting the `MICROBIT_USB_SERIAL_WAKE` config, when a battery is connected to the micro:bit, this programme wakes up when the USB cable is disconnected.

```cpp
#include "MicroBit.h"

MicroBit uBit;

int main() {
    uBit.init();
    uBit.power.deepSleep();

    while (true) {
        uBit.display.print("A");
        uBit.sleep(500);
    }
}
```
[MICROBIT.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/9627387/MICROBIT.hex.zip)

Not only that, but in some cases when on battery power (but not when PC USB connected), the board doesn't stay asleep. Pressing the reset button restarts the programme (the interface issues a restart command via SWD), and the programme will  permanently show the letter A on the display. It looks like it goes to sleep and immediately wake up (or never go to sleep, hard to tell).

Interestingly, if all power sources are removed, and then a battery is connected, the initial state of the board is to stay asleep, but if you press the reset button enough times the board sometimes doesn't stay asleep either.

I cannot replicate when I remove/disable the `MICROBIT_USB_SERIAL_WAKE` config.

JohnVidler, 2022-10-17:
At the moment, the changes to cause wake-on-uart-event is simply keeping the periperhal alive during a deep sleep; at the moment its unclear if this wake is caused by a bogus data transmission from DAPLink, OR if this is something being mishandled in CODAL.

I'll investigate logging the raw serial events so we can examine this, and work out how best to handle it.

microbit-carlos, 2023-01-26:
One possibility is that if the UART idle state is high and DAPLink goes to sleep (battery+USB power = awake, only battery = sleep) it might be pulling down the UART signal.