lancaster-university/codal-microbit-v2 #236: New audio.levelSPL returns fairly high values in a very quiet room.

microbit-carlos, 2022-09-22:
Using the new `uBit.audio.levelSPL` implemented in MicroPython (https://github.com/microbit-foundation/micropython-microbit-v2/commit/c5ab62716a72147893f8e3672f3f5fb177925e1f) returns fairly high values in a very quiet room:

```
37
18
32
32
42
18
32
32
26
```

Current MakeCode and older MicroPython betas with CODAL v0.2.40 return zeros in the environment and with the same micro:bit.
- [makecode-sound-to-serial.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/9629380/makecode-sound-to-serial.hex.zip)
```js
basic.forever(function () {
    serial.writeNumber(input.soundLevel())
    serial.writeLine("")
})
```
- [micropython-sound-to-serial.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/9629385/micropython-sound-to-serial.hex.zip)
```python
from microbit import *

while True:
    print(microphone.sound_level())
    sleep(150)
```

microbit-carlos, 2022-09-22:
@finneyj could you double check if you can spot anything odd with the MicroPython implementation?
- https://github.com/microbit-foundation/micropython-microbit-v2/commit/c5ab62716a72147893f8e3672f3f5fb177925e1f
- https://github.com/microbit-foundation/micropython-microbit-v2/blob/c5ab62716a72147893f8e3672f3f5fb177925e1f/src/codal_app/microbithal_microphone.cpp

It leaves everything to the defaults, simply inits with `uBit.audio.levelSPL->setUnit(LEVEL_DETECTOR_SPL_8BIT)` and then calls `uBit.audio.levelSPL->getValue()`. 

I certainly would not expect `uBit.audio.levelSPL->getValue()` to be returning values as high as 40-something in a silent room.

microbit-carlos, 2022-09-22:
A sample to replicate in CODAL:

```cpp
#include "MicroBit.h"

MicroBit uBit;

int main() {
    uBit.init();
    uBit.audio.levelSPL->setUnit(LEVEL_DETECTOR_SPL_8BIT);

    while (true) {
        int level = uBit.audio.levelSPL->getValue();
        uBit.serial.printf("%d\n", level);
        uBit.sleep(150);
    }
}
```
[MICROBIT.hex.zip](https://github.com/microbit-foundation/micropython-microbit-v2/files/9629508/MICROBIT.hex.zip)

finneyj, 2022-09-23:
Thanks @microbit-carlos.

Silence is relative :)  

The SPL level detector is now more sensitive at the lower end than it used to be. The sorts of values you shared above (in the 30-40 range) would correspond to about 40db-45db... a quiet but not silent room ;)

The 8 bit output is simply the sound pressure level in dB, mapped into a linear 8 bit 0..255 range. The start and end points in dB where that range is mapped are controllable by CONFIG options:

https://github.com/lancaster-university/codal-core/blob/master/inc/streams/LevelDetectorSPL.h#L53
https://github.com/lancaster-university/codal-core/blob/master/inc/streams/LevelDetectorSPL.h#L58

i.e. a default build will map 35dB or less to a value of 0, and 100dB or more to a value of 255. I chose 35db as that's the minimum level we can sense with microbit.

Before the recent patch, the lowest level we could sense was about 52dB... so if you'd like to maintain parity, we could just set LEVEL_DETECTOR_SPL_8BIT_000_POINT to 52. Anything less would then return zero.

Perhaps worth trying that in your test? If that feels more consistent, we can update the default in the codal-microbit-v2 for consistency across platforms?

finneyj, 2022-09-23:
p.s. It looks like you're seeing a fair bit of quantization down there too (looking at the data output). I didn't notice that in previous tests... it might be worth us adding a little smoothing on the output in due course too.

microbit-carlos, 2022-09-23:
Thanks @finneyj!

Okay, so building MicroPython with the setting `LEVEL_DETECTOR_SPL_8BIT_000_POINT` to `52.0f` and in my "quiet room" baseline we do get zeros 🎉 

We are getting better results with the new implementation than we did before, but any user programme using "quiet" and "loud" thresholds or events might needs to update the values.
Taking in consideration the improved results of the new level detector (better detection, or range of values, for the not-so-loud noises like speech and music), we think this is worth shipping this version.

finneyj, 2022-09-23:
Cool. Let's work the new default zero level into a new tag, so we keep a clean release for you.

closed 2022-10-11 by microbit-carlos