lancaster-university/codal-microbit-v2 #237: IDs MICROBIT_ID_MBED_* have been removed but they are not unused

martinwork, 2022-10-03:
IDs MICROBIT_ID_MBED_* have been removed by

Removed unused MBED IDs
https://github.com/lancaster-university/codal-microbit-v2/commit/53c1038fe37884edaf4ff86092398a728fd96cd8

but I think they may be used by some programs

https://github.com/lancaster-university/codal-microbit-v2/blob/master/inc/compat/InterruptIn.h#L25
https://github.com/lancaster-university/codal-microbit-v2/blob/master/inc/compat/PwmOut.h#L14
https://github.com/lancaster-university/codal-microbit-v2/blob/master/inc/compat/Timeout.h#L37
https://github.com/lancaster-university/codal-microbit-v2/blob/master/inc/compat/Ticker.h#L36

JohnVidler, 2022-10-17:
One option for this is to move the defines to a 'safer' space, as at the moment their values collide with other CODAL IDs elsewhere (the audio splitter, for example).

I'll check with @finneyj to see where we should move these, then reinstate them somewhere safe.

JohnVidler, 2022-11-04:
Ok, so I've moved these IDs to a 'safe' reserved ID space, and have them marked as deprecated.

The compiler will now warn you if you're using them in a program.

martinwork, 2022-11-05:
Arising from support ticket https://support.microbit.org/helpdesk/tickets/49545 (private)

@JohnVidler Where are they defined?

JohnVidler, 2022-11-06:
They're going to be included as part of the next tag, but I can push the changes for that repo now if there's a pressing need.

I wasn't aware of that specific support issue, apologies.

Do note, however, that using MBED specific stuff isn't really supported, they aren't considered part of the 'safe/stable' interface to the hardware and will cause deprecation warnings in your builds.

JohnVidler, 2022-11-06:
Now pushed, with the commit you want as https://github.com/lancaster-university/codal-microbit-v2/commit/85001d24a0109afb1a27f5e92ee216f5e444a081 pull `codal-microbit-v2` on `master` to get the changes.

martinwork, 2022-11-06:
Thanks @JohnVidler The support ticket is new since I noticed the IDs had been removed, and is from someone using the mbed online compiler, wanting to easily get their V1 code running on a V2.
I'm not using them, but I knew those IDs were used in the compatibility classes, which weren't removed, and thought there could be code that hasn't been updated since using the compatibility classes. I think all references to these IDs already generated warnings, like https://github.com/lancaster-university/codal-microbit-v2/blob/master/inc/compat/Ticker.h#L53.
In fact, I think I have seen those warnings in the Chrome console recently, but I can't remember when! 
Maybe CODAL Ticker should have used MESSAGE_BUS_LISTENER_IMMEDIATE though?

closed 2022-11-06 by martinwork

JohnVidler, 2022-11-07:
No prob - I had the fix ready to go but didn't realise it was pressing.

The `#warning` syntax works nicely with block-ifdef'd stuff like `Ticker` but having actual symbols and `__attribute__ ((deprecated))` tagging them also catches cases where folks attempt to listen for events with these IDs, even if the events are not possible.

Note also that the IDs have changed for those 4 values, and now no-longer collide with current CODAL IDs. (This was the cause of spurious audio pipeline wake events in other issues)

If folks need runtime dynamic IDs for cases where multiple instances of the same component exist and need to be distinguished, there will shortly be a `CodalComponent::generateDynamicID()` static method to generate safe (non-colliding) IDs properly.

martinwork, 2022-11-07:
> If folks need runtime dynamic IDs for cases where multiple instances of the same component exist and need to be distinguished, there will shortly be a CodalComponent::generateDynamicID() static method to generate safe (non-colliding) IDs properly.

Good idea!