lancaster-university/codal-microbit-v2 #240: `using namespace codal` shouldn't be used in header files

carlosperate, 2022-10-11:
There are a few header files in the different CODAL repositories that have a `using namespace codal;` line, which forces the namespace into anything that includes it and can cause ambiguity and clashes when integrating with other libraires/SDKs (kind of defeats the point of using a codal namespace to begin with).

I can see maybe a need to have it for `MicroBit.h`, but if we could remove its usage in all the other header files that would be useful.

finneyj, 2022-10-11:
yes, indeed. I recall this was a necessary evil to get codal-microbit-v2 out before launch. Times were tight!

Correct fix is the explicitly namespace the necessary types in header files to avoid namespace pollution...

finneyj, 2022-10-11:
p.s. I also agree MicroBit.h is a special case, and leaving one in there does make sense I think, in the interesting of keep user programs simpler and not making breaking changes.

finneyj, 2022-10-11:
maybe wrapped in a CONFIG option though, so the user can easily opt out should they wish to.

carlosperate, 2022-10-11:
> maybe wrapped in a CONFIG option though, so the user can easily opt out should they wish to.

Oh yeah, that'd be great!

microbit-carlos, 2024-07-24:
Fixed with:
- https://github.com/lancaster-university/codal-core/pull/170
- https://github.com/lancaster-university/codal-core/pull/171
- https://github.com/lancaster-university/codal-nrf52/pull/43
- https://github.com/lancaster-university/codal-nrf52/commit/660ee5a8499b69f2c92c2fe401f8bfe3e4fa302f
- https://github.com/lancaster-university/codal-nrf52/pull/54
- https://github.com/lancaster-university/codal-microbit-v2/pull/437

closed 2024-07-24 by microbit-carlos