lancaster-university/codal-microbit-v2 #242: Current CODAL builds are broken

microbit-carlos, 2022-11-09:
To replicate:
- clone microbit-v2-samples
- `python build.py`
- flash the MICROBIT.hex file
- Nothing scrolls on the display

Going into the libraries/codal-microbit-v2 repository and checking out the latest tag (5785bb077286faec8d8100f1ddf50d18140c115c), and then building again works.

microbit-carlos, 2022-11-09:
Don't know if this is related, but if the `SOFTDEVICE_PRESENT` flag can be set to zero, then all the `#ifdef SOFTDEVICE_PRESENT` in this commit should probably be `#if SOFTDEVICE_PRESENT`, no?
https://github.com/lancaster-university/codal-microbit-v2/commit/79ced4ec69fd03d3a5b3923aca18cf20bc511257

JohnVidler, 2022-11-09:
Hmm, I've been looking into this further and Carlos is correct, it does seem to be `SOFTDEVICE_PRESENT` breaking the build somewhere. Because many parts of the code use the flag _very early_ in the build, its a bit of a pain to shim some kind of check early on for another flag name, so I'm afraid it might be the case that we need to replace all those `#ifdef`s with `#if`s

finneyj, 2022-11-09:
I don't think we can replace the ifdefs in the the nrf-sdk / nrfx though @JohnVidler? In which case, would we not be better off using a different CONFIG option as carlos suggested?

JohnVidler, 2022-11-10:
I've now got this working locally, via some CMAKE ninja fu!

I'll be adding a new bit of syntax for `config: {}` blocks in `codal.json` whereby we can now specify configuration in the form `SET_FLAGNAME` and it will resolve to `-DFLAGNAME` on the build, provided that `SET_FLAGNAME` == 1. So in this particular case, the following configuration will _enable_ softdevice builds:

```
...
"config": {
   ...
   "SET_SOFTDEVICE_PRESENT": 1
   ...
}
```

... and the following will disable softdevice builds:

```
...
"config": {
   ...
   "SET_SOFTDEVICE_PRESENT": 0
   ...
}
```

The target can set a a default via the same mechanism as we currently have for other configuration data, and can be overidden by the syntax above in the projects, as required.

Note that as this changes the project `CMakeFiles.txt` to ensure that the flags are set early enough in the build, other downstream applications will need to update their respective `CMakeFiles.txt` to get this functionality.

In the near future, we plan to extend what we've built with `build.py` to include automatic updates for cases like this, whereby the tooling is decoupled from the project repositories. But for now, we need to deal with this manual step.

microbit-carlos, 2023-01-06:
@JohnVidler before closing this, as the bootstrap adoption should be around the corner, could you make sure any changes to `microbit-v2-samples` is applied to the upstream `codal` repo as well?

microbit-carlos, 2023-01-27:
Fixed in 0.2.44

closed 2023-01-27 by microbit-carlos