lancaster-university/codal-microbit-v2 #244: Possible sources of ID clashes, especially with jacdac

martinwork, 2023-01-06:
A range of IDs is reserved for jacdac...

https://github.com/lancaster-university/codal-core/blob/master/inc/core/CodalComponent.h#L83
// jacadac reserved from 3000 - 4000
#define DEVICE_ID_JD_DYNAMIC_ID 3000

Are there clashes with jacdac in these IDs above 3000?

https://github.com/lancaster-university/codal-microbit-v2/blob/4f76deafe67268227b7b4b9c2f471595e91ab2a5/model/MicroBit.h#L318

https://github.com/lancaster-university/codal-microbit-v2/blob/111e4cf829fdfea25e301375643d1bad71fb7070/inc/SoundEmojiSynthesizer.h#L46

https://github.com/lancaster-university/codal-microbit-v2/blob/111e4cf829fdfea25e301375643d1bad71fb7070/inc/Mixer2.h#L42

Beware IDs not defined with all the others...

https://github.com/lancaster-university/microbit-dal/blob/602153e9199c28c08fd2561ce7b43bf64e9e7394/inc/bluetooth/ExternalEvents.h#L33
https://github.com/lancaster-university/codal-microbit-v2/blob/111e4cf829fdfea25e301375643d1bad71fb7070/inc/bluetooth/ExternalEvents.h#L33

MakeCode has used IDs above 4000

https://github.com/microsoft/pxt-common-packages/blob/58f1ea42b83f09e1c76ccfe637cbc90b247b0fc3/libs/core/pxt.h#L115
#define DEVICE_ID_BUTTON_SLIDE 3000
#define DEVICE_ID_MICROPHONE 3001
#define DEVICE_ID_FIRST_BUTTON 4000
#define DEVICE_ID_FIRST_TOUCHBUTTON 4100

#define PXT_INTERNAL_KEY_UP 2050
#define PXT_INTERNAL_KEY_DOWN 2051

https://github.com/microsoft/pxt-common-packages/blob/3e54bbf9805405dd2da1930c6fea68867f59b9ce/libs/pulse/pulse.h#L15
#define PULSE_IR_COMPONENT_ID 0x2042             [8258]
#define PULSE_CABLE_COMPONENT_ID 0x2043      [8259]

_Originally posted by @martinwork in https://github.com/lancaster-university/codal-microbit-v2/issues/96#issuecomment-898884793_

microbit-carlos, 2023-01-19:
@finneyj to sync with @jamesadevine to see if we can reuse the IDs.

jamesadevine, 2023-01-19:
Would probably be better to sync with @pelikhan or @mmoskal on this!