lancaster-university/codal-microbit-v2 #247: Radio enable() and setSleep() do not retain power and frequency band settings

martinwork, 2023-01-17:
The defaults are used here:
https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/MicroBitRadio.cpp#L281

If that is the right for enable(), setSleep() needs to save and restore here:
https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/MicroBitRadio.cpp#L563

See also https://github.com/lancaster-university/codal-microbit-v2/issues/245

microbit-carlos, 2023-01-19:
@JohnVidler could you list here how you'd like this implemented and then @martinwork can create a PR?

JohnVidler, 2023-01-19:
From our discussions, it sounds like we need to:
- Track any radio register values (frequency, band, etc.) in member variables
- Pull the `enable()` functionality into an internal `reconfigure()` call
  - Reconfigure should check against the radio registers vs. our member variables and update/restart the radio if required
- Update any set*** calls to call our new `reconfigure()`

Unless this proves to be impractical.

martinwork, 2023-01-21:
PR https://github.com/lancaster-university/codal-microbit-v2/pull/252

@JohnVidler Changes to TXPOWER probably don't need a restart from setTransmitPower(), because the radio is restarted as part of the next send(), so the new restart code is only needed in setFrequencyBand. Would you like it in a function anyway?

JohnVidler, 2023-01-25:
That should be fine then, just in `setFrequencyBand` for now, but if we extend ths functions of the radio further and find we need to reconfigure elsewhere, then cycling the radio should be its own call.

Thanks for that @martinwork - I'll merge the PR in.

closed 2023-01-25 by JohnVidler