lancaster-university/codal-microbit-v2 #251: `wasTouched()` method for pins 0, 1, and 2

microbit-carlos, 2023-01-19:
We'd like to add a `was_touched()` method to the MicroPython `MicroBitTouchPin` class (like the `MicroBitButton` has), which has instances `pin_logo`, `pin0`, `pin1`, and `pin2`.

For micro:bit V2, `uBit.logo` is a `TouchButton` which has the method already and is all good, but pins 0, 1, 2 are instances of `NRF52Pin` which has a `isTouched()` method, but not a `wasTouched()`.

What would be the best way to add a MicroPython `was_touched()`  for pins 0, 1, and 2?

dpgeorge, 2023-02-13:
> the simplest way might be to add a `wasTouched()` method to `NRF52Pin`.

This would be good.  Then `uBit.io.logo` would also gain this new method (because it's a `NRF52Pin`), and that would make the MicroPython implementation easier.

dpgeorge, 2023-02-13:
And I think that `wasTouched()` should return the number of touches since the last call to that method, to match the semantics of `wasPressed()`.

JohnVidler, 2023-11-01:
This is now implemented in `master`, so should be in the next tag.

closed 2023-11-01 by JohnVidler

dpgeorge, 2023-11-13:
I think `wasPressed()` was only added to the `NRF52Pin` object, but not to the `TouchButton()` class?

JohnVidler, 2023-11-15:
Yes, to align with `NRF52Pin::isPressed`

`TouchButton` objects are only used internally within `NRF52Pin`s and the `uBit` object exposes all pins, edge connector or otherwise as Pin objects: https://github.com/lancaster-university/codal-microbit-v2/blob/master/model/MicroBitIO.h#L370 so every Pin gains the new method 🙂

microbit-carlos, 2023-11-17:
Moving the conversation to:
- https://github.com/lancaster-university/codal-microbit-v2/issues/396