lancaster-university/codal-microbit-v2 #253: Display animation ordering with concurrent access:  policy and consistency

finneyj, 2023-01-26:
We currently have a divergence of behaviour been the micro:bit v2 behaviour and MakeCode simulator behaviour for the following code example:

[image]

MakeCode simulator will interleave the access to the display animations, such that the "scroll" operation will occur in between any of the other "show icon" operations in the forever loop.

Following the recent patch here, codal will only schedule the "scroll" operation at the end of the forever loop. i.e. if  fiber is holding the lock on the display, it gets "first dibs" on reacquiring that lock before other fibers (including event handlers) can access it. https://github.com/lancaster-university/codal-core/commit/75b92a9ef3a206cdc63f65576c03e2aa72b4a45b

Looking at the CODAL source code, it is trivial to change the physical device's behaviour to align with the MakeCode simulator (and consistent with how fibers are normally scheduled to share access to CPU). However, it also looks a little like the current CODAL behaviour may have been intentionally introduced, most likely such that programs like this do not interleave animations:

[image]

So discussion points: 

1) Do we want to align behaviours of CODAL and MakeCode, or are we happy to leave the behaviour as undefined.
2) If we do want to align behaviours, what is the preferred behaviour? (the simulator behaviour is more consistent, the CODAL version is perhaps there to meet the needs of a specific UX use case).

@jaustin @microbit-carlos @martinwork

finneyj, 2023-03-09:
OK - based on the lack of discussion, there doesn't seem to be much appetite right now for introducing some sort of locking primitive in blocks languages. Which is fine.

As such, I've made the minor patch that aligns CODAL behaviour with the MakeCode simulator. Patch should be live in v0.2.50 tag: https://github.com/lancaster-university/codal-core/commit/434f12e52b9e551c96422ef76e6f9f3c5f6fa803

We can reopen discussion of what a "MUTEX block" would look like in the future if needed.

finneyj, 2023-03-09:
Right... following discussion on Lancaster/Foundation sync call today, I've reverted master branch to previous state to allow further discussion. https://github.com/lancaster-university/codal-core/commit/2dac8e2cb3b8fc02e7cb6ae0a3b50c5999fc2d49

@martinwork  @microbit-carlos  @jaustin @JohnVidler @abchatra @microbit-mark 
It feels like we're going around in circles about the behaviour we actually want for micro:bit when MakeCode programs have multiple fibers that interact with the display at the same time. Let's bring all the great minds together here and conclude once and for all on intended behaviour. :)

The question simply comes down to the behaviour we wish to define (or not) for the following code snippet:

[image]

Option 1: The current v1 and v2 DAL/CODAL behaviour allows fibers to monopolise the display until the stop interacting with it. Then until waiting fibers get the chance to access it. So the observed behaviours is a repeating pattern of:

- Big Heart
- Small Heart
- Smiley Face
- Surprised Faced

Option 2: MakeCode simulator takes a different approach, and interleaves each display call (which is also what the CODAL scheduler does in the general cases - fair load balancing and causal ordering among competing fibers). This results in a repeating pattern of:

- Big Heart
- Smiley Face
- Small Heart
- Surprised Faced

I've had various conversation with many of you, and the following facts can be noted:

1) We have a disparity between the simulated and actual behaviour for programs of this nature
2) We all want micro:bit v1 and v2 to behave the same in this regard
3) Option 1 seems more intuitive to kids and teachers 
4) Option 2 is more consistent with CODAL fiber semantics in the general sense (fair scheduling and causal ordering)
5) Aligning the MakeCode simulator with option 1 is "hard to fix from a timing perspective" (https://github.com/microsoft/pxt-microbit/issues/3870)
6) At least one member of foundation team think CODAL should align with the simulator ((https://github.com/microsoft/pxt-microbit/issues/3870)
7) At least one member of foundation team thinks that the simulator should align with CODAL. 

The way I see it I think we have three choices, and we just need to make a decision:

A) We adopt Option 1 above, and align the MakeCode simulator with CODAL
B) We adopt Option 2 above, and align CODAL with the MakeCode simulator
C) We declare the whole space of competing fibers as "undefined" and accept divergences in behaviour between v1, v2 and simulators.

Whilst I welcome discussion, the time for action is really upon us...

Can EACH of you please post here and indicate your preferred choice of behaviour... or raise a "fourth way" if you see one. :)

JohnVidler, 2023-03-09:
I vote for option 1, but only because the display as a 'resource' has a 'big lock' wrapping the `[set pixel][wait]` function pairs, so each block there is logically `[lock resource][set pixels][wait][unlock resource]`, and even through we would yield in the wait, any other running fibers are blocked on the resource lock and made to wait. Once all the rest are on the wait queue, we end up back in the original, which then unlocks the resource, and immediately re-enters a lock, with the same effect as before.
The only reason we ever see a switch at all is the effective double-yield that happens at the end of the repeat forever block via its implicit yield, so a complete op sequence looks like: `[lock][set pixels][wait][unlock][lock][set pixels][wait][unlock][yield]` with the last one allowing other threads the chance to get the display resource lock.

So I'd like to see Makecode reach parity with this, but I don't agree that your statement Joe that:
> Option 2 is more consistent with CODAL fiber semantics in the general sense (fair scheduling and causal ordering)

Because Option 1 _is_ doing this, just its being so completely fair that you don't notice :)

To summarise:
- I vote for Option 1
- Option 1 is most correct to the mode
- Option 1 is most correct to the concept

finneyj, 2023-03-09:
Thanks John. Fair treatment is always in the eye of the beholder. :)

martinwork, 2023-03-09:
I vote for Option 1, the current V1 regime.

I think Option 2 would be most faithful to the round robin fiber behaviour, but special casing the display with Option 1 makes some simple natural things easy, that would be quite difficult with Option 2. With Option 1, the interleaving of Option 2 can be enabled by inserting pauses.

finneyj, 2023-03-09:
Thanks @martinwork

JohnVidler, 2023-03-09:
I think that Option 2 would be round-robin'ing _resource access_ rather than _fiber execution_ which is a little different. Just normally these look like the same thing for low latency IO, unfortunately the display is effectively high-latency.

abchatra, 2023-03-10:
I fine with either option 1 or option 2 but be consistent between MakeCode and CODAL. 
If we are all going for option 1, we will evaluate the cost to fix it in the simulator from MakeCode .

jaustin, 2023-03-10:
I'd like option 1 please.

finneyj, 2023-03-10:
OK - thanks all. Looks like we're pretty quorate on Option 1... down goes the gavel! Let all who question it forever hold their peace! :)

Option 1 is the current behaviour we have in codal and microbit-dal, so I'll close this issue off shortly.

@abchatra - many thanks for offering to review. I've created an issue over on pxt-microbit so that we have some tracking after I close here: https://github.com/microsoft/pxt-microbit/issues/4982. Depending on how the sim has been implemented, it might be a very simple patch if it follows the codal asyc implementation (I think Peli did this for most peripherals), and the chaneg in CODAL was literally a one line patch: https://github.com/lancaster-university/codal-core/commit/2dac8e2cb3b8fc02e7cb6ae0a3b50c5999fc2d49

microbit-carlos, 2023-04-11:
As no other changes were required for option 1 this was "officially shipped" in v0.2.50 🎉

closed 2023-04-11 by microbit-carlos