lancaster-university/codal-microbit-v2 #257: Review and merge codal-microbit-nrf5sdk PRs

microbit-carlos, 2023-01-27:
As they are in the micro:bit Foundation org they cannot be added to the CODAL project board, so creating this issue to be able to track them:

- https://github.com/microbit-foundation/codal-microbit-nrf5sdk/pull/12
- https://github.com/microbit-foundation/codal-microbit-nrf5sdk/pull/13

martinwork, 2023-07-04:
When I tried the ble_test() sample just now the services and characteristics don't all fit. The UART characteristics are missing.

The first PR above was intended to fix that, but some caution is needed now because the build has changed since I made that PR.

martinwork, 2023-07-09:
@microbit-carlos @JohnVidler I think the effect of not having https://github.com/microbit-foundation/codal-microbit-nrf5sdk/pull/13 is that when MakeCode is set to **no pairing required**, the build uses open security for everything except the Nordic DFU characteristic. The iOS app first writes to an open security characteristic, then fails to flash with the error **authentication is insufficient** for the Nordic DFU. A workaround is to use MakeCode Edit Settings As Text to define "security_level": "SECURITY_MODE_ENCRYPTION_OPEN_LINK" instead if the default **null**.

microbit-carlos, 2023-07-14:
So does that mean that when this option is set in MakeCode the micro:bit cannot be flashed via the apps?
[image]

martinwork, 2023-07-14:
@microbit-carlos I believe an open link hex from MakeCode can be flashed but can't be flashed over. Only tested in iOS. I dragged the the open link hex to the MICROBIT drive, then tried to pair with and BLE flash over it with the iOS app. 

It may be that if you have paired with another hex, then BLE flashed an open link hex, the pairing will allow another hex to be flashed. I'm not sure exactly what MakeCode creates, but I think this is a likely explanation for the **authentication is insufficient** error.

microbit-carlos, 2023-07-18:
Is it safe to assume this was also a problem in MakeCode v5? The issue is not something introduced between v5 and v6?

martinwork, 2023-07-18:
Compare
https://github.com/microbit-foundation/codal-microbit-nrf5sdk/blob/master/sdk_config/sdk_config.h#L102
with 
https://github.com/lancaster-university/codal-microbit-v2/blob/master/inc/MicroBitConfig.h#L178
The latter is correct because it doesn't redefine MICROBIT_BLE_SECURITY_MODE when MICROBIT_BLE_OPEN has already defined it.

If I build microbit-v2-samples ble-test() with the codal.json below I can go through the pairing process with it in the app, then flash over it. [MICROBIT open good.zip](https://github.com/lancaster-university/codal-microbit-v2/files/12088135/MICROBIT.open.good.zip)

If I remove the last line from codal.json or change the value to null and rebuild, there are lots of warnings (warning: "MICROBIT_BLE_SECURITY_MODE" redefined) but it does build. Pairing with the result fails with Authentication Insufficient.
[MICROBIT open bad.zip](https://github.com/lancaster-university/codal-microbit-v2/files/12088137/MICROBIT.open.bad.zip)

I also get Authentication Insufficient pairing with this, saved from MakeCode v5  [microbit-openlink-v5 (8).zip](https://github.com/lancaster-university/codal-microbit-v2/files/12087872/microbit-openlink-v5.8.zip) https://makecode.microbit.org/_UMt84d9Cuiut

If I pair with a normal, MakeCode just works hex, and BLE flash an open hex over it,  sending another hex initially fails. If it was a "good" open hex, it's possible to recover by pairing again, but pairing with a "bad" open hex fails with Authentication Insufficient.

```
{
    "target": {
        "name": "codal-microbit-v2",
        "url": "https://github.com/lancaster-university/codal-microbit-v2",
        "branch": "master",
        "type": "git"
    } ,
    "config":{
        "DEVICE_BLE": 1,
        "MICROBIT_BLE_ENABLED" : 1,
        "MICROBIT_BLE_PAIRING_MODE": 1,
        "MICROBIT_BLE_DFU_SERVICE": 1,
        "MICROBIT_BLE_DEVICE_INFORMATION_SERVICE": 1,
        "MICROBIT_BLE_EVENT_SERVICE" : 1,
        "MICROBIT_BLE_PARTIAL_FLASHING" : 0,
        "MICROBIT_BLE_OPEN" : 1,
        "MICROBIT_BLE_SECURITY_LEVEL": "SECURITY_MODE_ENCRYPTION_OPEN_LINK"
    }
}
```