lancaster-university/codal-microbit-v2 #258: Setting the Power LED Sleep State OFF

HemanthSabbella, 2023-01-30:
I want to turn off the power LED for fine energy optimizations. [Power Management Specification](https://tech.microbit.org/software/spec-power-management/) states that when the Power LED Sleep state setting is off, the interface KL27 turns the power LED off. Could anyone please help me change the default setting of the Power LED Sleep state?

microbit-carlos, 2023-02-16:
Hi @HemanthSabbella,

When target MCU sets the `Power LED Sleep` to `ON`, the red LED will always be light up. This is because the target MCU is running and the power LED should indicate so.

> When the `Power LED Sleep` state setting is `OFF`, the Interface (KL27) must turn OFF the Power (red) LED when it goes into Sleep or Power Down mode

And then, when the target goes to sleep it will set the `Power LED Sleep` state to `OFF`. In essence at this point it passes control of the LED to the Interface chip. This is because while the Interface MCU is running the power LED should be light up, to indicate one of the chips on the board is still awake. And when the Interface MCU finally goes to sleep (when it's not  maintaining a USB connection) it will then turn off the LED.

If the red LED is ON when the target is sleeping and the board is on battery power, then it means the Interface MCU is still running for some reason. Is that what you are experiencing?

microbit-carlos, 2023-03-29:
Hi @HemanthSabbella, I hope that answered your question, so I'll close this as resolved.
But if it didn't, or you have any other question, please feel free to reopen the issue 👍

closed 2023-03-29 by microbit-carlos