lancaster-university/codal-microbit-v2 #259: API for setting/requesting wall-clock time

jaustin, 2023-01-30:
With increased prevalence of data logging features it would be really good to have a well codified way of setting time programatically.

There are a few 'sources' of time:
* A BLE service like the one mentioned in #178
* An accessory like a GPS chip
* web USB (via a custom protocol)
* Serial

And then we may want to 'save' this time:
* using an external RTC
* using the target device RTC
* using the interface chip RTC (would require new DAPLink version @microbit-carlos )

I would like us to think about the way this stuff should intersect with user code and accessory add-ons

jaustin, 2023-01-30:
CC @bsiever and @DaveAtKitronik I think you both will have an interest in this topic

bsiever, 2023-01-30:
Some sort of wall-clock time setting could make a lot of activities a lot easier (clocks & smart watches, data logging, etc.).  My less-than-perfect software-based RTC (https://github.com/bsiever/microbit-pxt-timeanddate) would benefit from a simpler approach to setting time too. 

The big downside of BLE is that it's mutually exclusive with radio.  

I took a quick look at the CMSIS-DAP commands and didn't see a RTC-related command (could use a second look --- but I wouldn't really expect support for RTC).  Maybe a Vendor CMSIS-DAP command?  And some support in the common MB environments that would check for it and set time if it were supported?   Battery-backed RTCs would be ok with a set-when-programmed approach, but it may not work well for the software-based RTC.

I suspect most users just want local time, but it's probably good to consider if UTC or keeping track of the UTC offset should be included.

DaveAtKitronik, 2023-01-30:
we already have https://github.com/KitronikLtd/pxt-kitronik-rtc-base and https://github.com/KitronikLtd/micropython-microbit-kitronik-rtc for the RTC chips we use on Kitronik products. Maybe something useful in there for starting points.

bsiever, 2023-01-30:
@DaveAtKitronik I think @jaustin is suggesting some approach to allow the time to be set directly to the RTC rather than programmatically.  For example, a MakeCode user would either set the time via MakeCode or it would be set automatically when code is transferred.   

Requiring the user to create code that will set the clock to a future time, program the code, and then trigger the code in synchronization with the real time is a little awkward for some users.   I think it's a nice "thinking about computers/programming" exercise, but it's awkward for those who just want to collect data and know the wall-clock time of the data points.

@jaustin Maybe a new Microbit event for time set and a "mailbox" (flash location?) that represents the last value it was set to (because events can't hold more than 16-bits of data)?  Any RTC (or software RTCs) that want to use it would just listen for the event and do whatever transformations they needed on the provided time.  Perhaps the provided time would be: 1) current UTC timestamp and 2) Estimated offset of UTC for current local?   Then the time APIs could decide which they want to use.  

A custom DAP command or BLE service could both trigger the event.