lancaster-university/codal-microbit-v2 #260: Clarification on Transmit Power Values

HemanthSabbella, 2023-01-31:
The setTranmistPower allows you to set the TX power from 0-7, corresponding to -30dBm to +4dBm. The source code surprisingly has ten levels, though the documentation only has 0-7 levels [(link](https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/MicroBitRadio.cpp)).

Further, the nRF52833 states that the lowest programmable limit is -20 dBm ([link](https://www.nordicsemi.com/-/media/Software-and-other-downloads/Product-Briefs/nRF52833-SoC-Product-Brief.pdf)). The microbit: firmware documentation ([link](https://lancaster-university.github.io/microbit-docs/ubit/radio/)) mentioned power = 0 is equivalent to -30dBm. 

The mapping is unclear and needs clarification.

microbit-carlos, 2023-02-16:
Thanks for raising this @HemanthSabbella.

Having a look, we have for DAL: [microbit-dal/source/bluetooth/MicroBitBLEManager.cpp#L80](https://github.com/lancaster-university/microbit-dal/blob/602153e9199c28c08fd2561ce7b43bf64e9e7394/source/bluetooth/MicroBitBLEManager.cpp#L80)
```cpp
const int8_t MICROBIT_BLE_POWER_LEVEL[] = {-30, -20, -16, -12, -8, -4, 0, 4};
```

With:
[microbit-dal/inc/drivers/MicroBitRadio.h#L67](https://github.com/lancaster-university/microbit-dal/blob/602153e9199c28c08fd2561ce7b43bf64e9e7394/inc/drivers/MicroBitRadio.h#L67)
[microbit-dal/source/drivers/MicroBitRadio.cpp#L296](https://github.com/lancaster-university/microbit-dal/blob/602153e9199c28c08fd2561ce7b43bf64e9e7394/source/drivers/MicroBitRadio.cpp#L296)
```cpp
#define MICROBIT_RADIO_DEFAULT_TX_POWER         6
```
So the there are 8 levels, default power would be 0 dBm, min -30, max 4dBm.

And for CODAL:
https://github.com/lancaster-university/codal-microbit-v2/blob/bf0bb964fbacbc44bf2a1faff09c211a47a1285b/source/MicroBitRadio.cpp#L35
```cpp
const uint8_t MICROBIT_RADIO_POWER_LEVEL[] = {
  0xD8,  // -40 dBm
  0xD8,  // -40 (technically 0xE2 is the datasheet value, but it's deprecated and sets the power to -40 anyway)
  0xEC,  // -20 dBm
  0xF0,  // -16 dBm
  0xF4,  // -12 dBm
  0xF8,  // -8 dBm
  0xFC,  // -4 dBm
  0x00,  // 0 dBm
  0x03,  // 3 dBm
  0x04   // 4 dBm
};
```

https://github.com/lancaster-university/codal-microbit-v2/blob/bf0bb964fbacbc44bf2a1faff09c211a47a1285b/inc/MicroBitRadio.h#L72
https://github.com/lancaster-university/codal-microbit-v2/blob/bf0bb964fbacbc44bf2a1faff09c211a47a1285b/inc/MicroBitRadio.h#L77

So there are 10 levels, default power would be 0 dBm as well, min -40, and max 4 dBm.

The CODAL docstrings do indicate the input range is 0 to 7, even though the check is done for values 0 to 9:
https://github.com/lancaster-university/codal-microbit-v2/blob/bf0bb964fbacbc44bf2a1faff09c211a47a1285b/source/MicroBitRadio.cpp#L123-L133
[image]

Ultimately, for compatibility with DAL and MakeCode (as MakeCode needs to set a value range that works in both DAL and CODAL) it makes sense to keep the CODAL range from 0 to 7 as well.

Otherwise code that has been written to use the max power available (`uBit.radio.setTransmitPower(7)`) would be setting the power to 0 dBm in V2.

Apart from that, is not very intuitive that in V2 `setTransmitPower(0)` and `setTransmitPower(1)` would produce the same -40 dBm power setting.

We can drop one of the`0xD8 (-40 dBm)` and the `0x03 (3 dBm)` values, to match DAL power values.

microbit-carlos, 2023-02-16:
@finneyj was there any reason to increase the value range for `MicroBitRadio::setTransmitPower()` in V2?

I assume the change proposed above would benefit MakeCode and not break any other dependents? (MicroPython is using its own radio driver with input values 0 to 7)

microbit-carlos, 2023-09-04:
Fixed in https://github.com/lancaster-university/codal-microbit-v2/pull/275 and will be released in tag v0.2.60.

closed 2023-09-04 by microbit-carlos