lancaster-university/codal-microbit-v2 #265: Clap detection

microbit-carlos, 2023-02-09:
Similarly how we can detect "loud" or "quiet" noises, we want to be able to identify claps more accurately (not just as loud noises).

JohnVidler, 2023-03-14:
Oliver has pushed his changes to the code repo now, so we should be able to merge this in for 0.2.50. I just need to consolidate the changesets and check his changes still work alongside demand activation.

olihulland, 2023-03-15:
Currently, the behaviour of this implementation means that it is possible for a Loud event and a Clap event to be emitted due to the same noise. This is because the loud event has a configurable threshold that may or may not be met during a clap and the determination of a clap works slightly differently to simply reaching a set level.

Need to consider whether we would want the clap event to inhibit the loud event in this case. I'm leaning towards no since this seems the most natural to me at first glance but I suppose this needs to be consistent with the wider behaviours.

JohnVidler, 2023-03-23:
I've heard no objections to claps _also_ being onLoud events so lets go ahead with this as-is.

Merged in as of 99b9399857fd5e6110c2f0b9f2b3d134cd59e92d 👍

closed 2023-03-23 by JohnVidler

microbit-carlos, 2023-03-24:
Link to commit: https://github.com/lancaster-university/codal-core/commit/99b9399857fd5e6110c2f0b9f2b3d134cd59e92d