lancaster-university/codal-microbit-v2 #269: uBit.audio.levelSPL->getValue() always returns zero

martinwork, 2023-02-09:
With the program below, and latest master, uBit.audio.levelSPL->getValue() always returns zero.

If I checkout v0.2.40, the values are between 52 and 87.

```
#include "MicroBit.h"

#ifndef MICROBIT_CODAL
#ifdef CODAL_CONFIG_H
#define MICROBIT_CODAL 1
#else
#define MICROBIT_CODAL 0
#endif
#endif

MicroBit uBit;

void onButtonA(MicroBitEvent e)
{
    uBit.audio.levelSPL->setHighThreshold( 120.0);
    uBit.audio.levelSPL->setLowThreshold( 52.0);
}

void onButtonB(MicroBitEvent e)
{
    uBit.audio.levelSPL->setHighThreshold( 85.0);
    uBit.audio.levelSPL->setLowThreshold( 65.0);
    
}

void onLoud( MicroBitEvent)
{
    uBit.serial.send("LOUD\n");
}

void onQuiet(MicroBitEvent)
{
    uBit.serial.send("QUIET\n");
}

void forever2()
{
    uBit.display.scroll("00");
    uBit.audio.enable();
    uBit.audio.activateMic();

    while (true)
    {
      float soundLevel = uBit.audio.levelSPL->getValue();
      ManagedString str( (int) floor( 0.5 + soundLevel));

      uBit.serial.send( str + "\n");
      uBit.sleep(200);
    }
}

void togglePixel( int x, int y)
{
    uBit.display.image.setPixelValue( x, y, uBit.display.image.getPixelValue( x, y) ? 0 : 255);
}

void forever1()
{
    while (true)
    {
      togglePixel( 0, 0);
      uBit.sleep(1000);
    }
}

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();

    create_fiber( forever1);
    create_fiber( forever2);
    
    uBit.messageBus.listen( MICROBIT_ID_BUTTON_A,  MICROBIT_BUTTON_EVT_CLICK, onButtonA);
    uBit.messageBus.listen( MICROBIT_ID_BUTTON_B,  MICROBIT_BUTTON_EVT_CLICK, onButtonB);

    uBit.messageBus.listen(DEVICE_ID_MICROPHONE, LEVEL_THRESHOLD_HIGH, onLoud);
    uBit.messageBus.listen(DEVICE_ID_MICROPHONE, LEVEL_THRESHOLD_LOW, onQuiet);

    // If main exits, there may still be other fibers running or registered event handlers etc.
    // Simply release this fiber, which will mean we enter the scheduler. Worse case, we then
    // sit in the idle task forever, in a power efficient sleep.
    release_fiber();
}

```

microbit-carlos, 2023-02-10:
Oh, btw @martinwork, could you test the "set threshold" issue with CODAL v0.2.40 in the meantime? That way we can look into these two issues in parallel.

microbit-carlos, 2023-02-10:
Also, for this issue, if you checkout CODAL v0.2.43 does it work?
v0.2.43 is the version currently used in MicroPython, which shipped with a working levelSPL:
https://github.com/microbit-foundation/micropython-microbit-v2/blob/v2.1.1/src/codal_app/microbithal_microphone.cpp

jaustin, 2023-02-10:
Cool. I was using the live version of MakeCode where I found the issue that setting the loud threshold didn't appear to do anything, so think we may *also* have a bug there, but can't deny this looks wrong too!

martinwork, 2023-02-10:
It was while trying to make C++ example for the threshold problem, that I discovered that the SPL value was broken. Took me quite a while to realise that it wasn't because I wasn't doing it right!

Here's a MakeCode (live, CODAL 0.2.40) example showing that setting the threshold high or low has no effect.
It uses the MakeCode console, to record on loud triggers (high), the sound level, and the loud threshold (loud).
https://makecode.microbit.org/_Ma29EU70DKKX
The MakeCode level never goes above 131, which maps to CODAL level 87. MakeCode clips the CODAL value to [52,120] and maps that range to [0,255].
The sample starts with threshold = 255 (~CODAL 120), but loud claps still trigger on loud.
Press A to set the threshold to 40. Then it's possible to record levels well above 40 without triggering on loud, but still possible to trigger with a clap.
Press B to set the threshold to 150 and see results like 255, just to confirm that 255 isn't a special case.

I'm working on a corresponding C++ sample. I'm seeing strange behaviour which seems to change depending on the order of calls... I'll be back!

martinwork, 2023-02-10:
Here's C++, with hex built with 0.2.40, with matching MakeCode console output. It uses CODAL level 120 to start, A= 65, B = 80.
[soundlevel-0240.zip](https://github.com/lancaster-university/codal-microbit-v2/files/10711081/soundlevel-0240.zip)
onLoud seems to trigger maybe once after a level change.
0.2.43 version to follow...

martinwork, 2023-02-11:
Built with 0.2.43, [test-soundlevel-v2-0243.zip](https://github.com/lancaster-university/codal-microbit-v2/files/10714331/test-soundlevel-v2-0243.zip) seems to trigger the on loud events as expected. 
The range of levels I see has changed from [52,87] in 0.2.40 to [33,103] in 0.2.43.

microbit-carlos, 2023-02-13:
Thanks @martinwork! Could you open a new GitHub issue for the setting the threshold problem? We should keep these two different problems separated or the threads end up being hard to follow.

microbit-carlos, 2023-02-13:
Btw, I can also confirm that getValue() is also returning zeros in MicroPython updated to the 0.2.48 CODAL release.

JohnVidler, 2023-02-13:
> I can confirm this is also returning only zeros in MakeCode with CODAL v0.2.48:
> 
> MakeCode test URL: https://makecode.microbit.org/app/62bd528c2aa51e6342a764c506492937d31ba568-a9bfe06ee6
> 
> Test code:
> 
> [image]
> 
> ```ts
> basic.forever(function () {
>     serial.writeLine("" + (input.soundLevel()))
> })
> ```

So I tried this on a 2.21 and it looks like the mic itself doesn't actually activate - can you confirm that you're actually seeing the mic LED light up? It might just be that we're not seeing the pipeline fire up for some blocks.

I'll continue to investigate.

JohnVidler, 2023-02-13:
OK, so this gets is where PXT is calling the audio subsystem: https://github.com/microsoft/pxt-microbit/blob/master/libs/microphone/microphone.cpp#L64 and it looks like it never manually calls any of the mic or audio activation calls, so we should probably do it ourselves...

The old level detector: https://github.com/lancaster-university/codal-core/blob/master/source/streams/LevelDetector.cpp#L111 and the new level detector: https://github.com/lancaster-university/codal-core/blob/master/source/streams/LevelDetectorSPL.cpp#L148 both seem to auto-activate themselves and attempt to connect to their upstream to trigger the upstream class, which in this case is a rawSplitter channel for both: https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/MicroBitAudio.cpp#L82

We create and connect a channel, then emit a `SPLITTER_ACTIVATE_CHANNEL` here: https://github.com/lancaster-university/codal-core/blob/master/source/streams/StreamSplitter.cpp#L198 which gets caught here, and forces a mic activation: https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/MicroBitAudio.cpp#L93 BUT StreamSplitters use a dynamic ID now as we have multiple of them at various layers (see https://github.com/lancaster-university/codal-core/blob/master/inc/streams/StreamSplitter.h#L92 ), so we need to collect the IDs and listen explicitly to these IDs here, not the constant def.

So, actions to fix are to just to collect the IDs of the splitters on create, then listen to those explicitly, and we should be sorted. I'll see if I can get a patch done and out for *.49, then we can test and hopefully close this :)

martinwork, 2023-02-13:
Why have the SPL levels changed from [52,87] in 0.2.40 to [33,103] in 0.2.43? MakeCode clips off vlaues below 52.

I made a simpler test with no activateMic() call (below), and confirmed that works in 0.2.40 and 0.2.43, but not 0.2.48 or master, as expected.

I realised I could use a MakeCode extension to confirm what MakeCode sees in the macros in https://github.com/microsoft/pxt-microbit/blob/master/libs/microphone/microphonehw.cpp#L42. This is what I got, as expected:
```
MIC_DEVICE = NRF52ADCChannel*
MIC_INIT = : microphone(uBit.audio.mic) , level(* uBit.audio.levelSPL)
MIC_ENABLE =
```

Some observations... mentioning them in case it helps!

StreamSplitter::createChannel() is only called from MicroBitAudio::MicroBitAudio, and it's called 3 times when MicroBitAudio::onSplitterEvent hasn't yet been registered.

Will there will be 3 SPLITTER_ACTIVATE_CHANNEL events queued up?

Why doesn't MicroBitAudio::MicroBitAudio just call activateMic() after hooking up the channels?

StreamSplitter::createChannel() can fire SPLITTER_ACTIVATE_CHANNEL even when the channel is not successfully added.

If `rawSplitter = new StreamSplitter` fails, it will panic, but if createChannel() returns NULL, `levelSPL = new LevelDetectorSPL(*rawSplitter->createChannel()` will crash.

SPLITTER_DEACTIVATE_CHANNEL is not fired anywhere.

```
#include "MicroBit.h"

MicroBit uBit;

void forever()
{
    while (true)
    {
      float soundLevel = uBit.audio.levelSPL->getValue();

      ManagedString str( (int) floor( 0.5 + soundLevel));
      uBit.serial.send( "level:" + str + "\n");
      uBit.sleep(200);
    }
}

int main()
{
    uBit.init();
    create_fiber( forever);
    release_fiber();
}

```

JohnVidler, 2023-02-16:
Nonetheless, I now have a fix for the issue here (ie. that the mic never activates for getLevel()) and that will be in `0.2.49`.

The audio pipeline is now properly aware of demand activations, and will bring the microphone up when a downstream component connects, and as the level detectors (both of them) demand-connect to their splitter channels, we can capture this and enable the mic when high-level languages (or CODAL programs) call any methods that prod the pipeline.

I'm considering this 'fixed' now, but we'll close this when we make the tag.

martinwork, 2023-02-17:
@JohnVidler Thanks! I was thinking I must be misunderstanding something. I'll run the tests again soon.

martinwork, 2023-02-19:
@JohnVidler The simple test in https://github.com/lancaster-university/codal-microbit-v2/issues/269#issuecomment-1428698944 still seems to fail. activateMic() doesn't seem to get called automatically when the level is read. The mic light doesn't come on. I tweaked the test so I could press A to call activate Mic - that works. I tried building it 3 ways! Am I failing to get the latest version of something?

Searching the source I am building, it makes sense to me that it doesn't work. As before:
- activateMic() is only called in onSplitterEvent for SPLITTER_ACTIVATE_CHANNEL
- SPLITTER_ACTIVATE_CHANNEL is only fired from createChannel()
- createChannel() is only called from MicroBitAudio::MicroBitAudio, before onSplitterEvent is registered.

I am still struggling to understand the streams code, but I guess the automatic mic activation should be kicked off when LevelDetectorSPL::getLevel calls connect on its upstream DataSource.
https://github.com/lancaster-university/codal-core/blob/master/source/streams/LevelDetectorSPL.cpp#L152

Previously the SPL upstream source was a StreamSplitter
https://github.com/lancaster-university/codal-microbit-v2/blob/v0.2.43/source/MicroBitAudio.cpp#L82
StreamSplitter::connect fired SPLITTER_ACTIVATE_CHANNEL
https://github.com/lancaster-university/codal-core/blob/0e0d4888f6cd75fd2588d507503b9065334c2763/source/streams/StreamSplitter.cpp#L135

Now the SPL upstream source is a SplitterChannel from StreamSplitter::createChannel().
https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/MicroBitAudio.cpp#L82
https://github.com/lancaster-university/codal-core/blob/master/source/streams/StreamSplitter.cpp#L180
SplitterChannel::connect doesn't fire SPLITTER_ACTIVATE_CHANNEL
https://github.com/lancaster-university/codal-core/blob/master/source/streams/StreamSplitter.cpp#L81

```
#include "MicroBit.h"

MicroBit uBit;

void onButtonA(MicroBitEvent e)
{
    uBit.audio.activateMic();
}

void forever()
{
    while (true)
    {
      float soundLevel = uBit.audio.levelSPL->getValue();

      ManagedString str( (int) floor( 0.5 + soundLevel));
      uBit.serial.send( "level:" + str + "\n");
      uBit.sleep(200);
    }
}

int main()
{
    uBit.init();
    create_fiber( forever);
    uBit.messageBus.listen( MICROBIT_ID_BUTTON_A,  MICROBIT_BUTTON_EVT_CLICK, onButtonA);
    release_fiber();
}
```

JohnVidler, 2023-02-27:
My apologies @martinwork  and anyone else who was confused by this, I wasn't clear about if the patches were in master on _here_ or locally ready to go. Entirely my fault and poor communication on my part.

I've now pushed and tagged 0.2.49, which _does_ include all the patches required to make this work, so if folks want to test against that now it should all be working.

0.2.49 is a candidate for a proper release, so if this is solved we will go ahead and do the release too.

martinwork, 2023-02-27:
Thanks @JohnVidler. The test runs fine now.